volkszaehler/vzlogger #577: Vzlogger GPIO Problem seit Umstellung auf neueste Raspi-Version #526 update

whymanda, 2023-02-08:
Habe das gleiche Problem reproduziert. Volkszähler-Image auf PI, sobald ich S0 aktiviere tritt der Fehler auf. Nach Reboot muss ich warten, irgrndwann nach 30-90 Sekunden kann ich vzlogger starten. S0-Impulse werden zwar erkannt aber nicht in die DB eingetragen. 1Wire funktioniert. 

[Feb 08 21:26:50][chn0] New channel initialized (uuid=...e44fa7 api=volkszaehler id=28-0b2280e55a71)
[Feb 08 21:26:50][chn1] New channel initialized (uuid=...a9c404 api=volkszaehler id=28-0b2280e6b180)
[Feb 08 21:26:50][main] log level is 5
[Feb 08 21:26:50]       Daemonize process...
[Feb 08 21:26:50][]     Startup failed: open direction failed
[Feb 08 21:27:09][main] vzlogger v0.8.1 based on heads/master-0-g76e868feed from Wed, 20 Apr 2022 08:39:54 +0200 started.
[

[Feb 08 21:39:38][chn0] New channel initialized (uuid=...e44fa7 api=volkszaehler id=28-0b2280e55a71)
[Feb 08 21:39:38][chn1] New channel initialized (uuid=...a9c404 api=volkszaehler id=28-0b2280e6b180)
[Feb 08 21:39:38][main] log level is 5
[Feb 08 21:39:38]       Daemonize process...
[Feb 08 21:39:38][mtr0] Meter connection established
[Feb 08 21:39:38][w1t]  open found 2 w1 devices
[Feb 08 21:39:38][mtr1] Meter connection established
[

J-A-U, 2023-02-09:
Hast du denn die in #526 vorgeschlagene Lösung implementiert?

whymanda, 2023-02-09:
Danke für die schnelle Nachfrage, ja beiden Installationen und mit mehrfachen Reboots. Für mich sieht es so aus, dass nach dem Reboot ein Thread/Process hängt, wenn man später vzlogger beendet und startet, gibt es das nicht mehr:

[Feb 09 12:19:47][mtr1] waiting 300 seconds before next reading
[Feb 09 12:23:20][]     main loop indicating all mappings to quit	<<<<<<<<<<<<
[Feb 09 12:23:26][main] vzlogger v0.8.1 based on heads/master-0-g76e868feed from Wed, 20 Apr 2022 08:39:54 +0200 started. <<<<<<<<<<<<<
[Feb 09 12:23:26][s0]   unknown option for mmap (). Falling back to normal gpio.
[Feb 09 12:23:26]       New meter initialized (protocol=s0)
[Feb 09 12:23:26]       New meter initialized (protocol=w1therm)
[Feb 09 12:23:26][chn0] New channel initialized (uuid=...e44fa7 api=volkszaehler id=28-0b2280e55a71)
[Feb 09 12:23:26][chn1] New channel initialized (uuid=...a9c404 api=volkszaehler id=28-0b2280e6b180)
[Feb 09 12:23:26][main] log level is 5
[Feb 09 12:23:26]       Daemonize process...
[Feb 09 12:23:26][mtr0] Meter connection established
[Feb 09 12:23:26][w1t]  open found 2 w1 devices
[Feb 09 12:23:26][mtr1] Meter connection established
[Feb 09 12:23:28][chn0] Adding reading to queue (value=18.44 ts=1675941808605)
[Feb 09 12:23:28][chn1] Adding reading to queue (value=1.69 ts=1675941807725)
[Feb 09 12:23:28][mtr1] waiting 300 seconds before next reading 

Bin übrigen es älterer Herr (habe Konrad Zuse noch persönlich 1969 kennengelern) und das sind meine ersten Gehversuche in einem Forum;-)

Whitytom, 2023-02-15:
Auch bei mir das gleiche Problem mit den aktuell verfügbaren Images vom 20.04.22 und 10.06.22. #526 war in Zweiterem schon verbaut, im Ersten hab ich es eingebaut -> keine Auswirkung.
Gibt es als schnellen Workaround irgendwo ältere Raspi-Images?

whymanda, 2023-02-17:
Ich habe das Image dreimal neu aufgesetzt, der Fehler bleibt. Habe unter root in die crontab ein keines Skript gebaut und prüfe alle zwei Minuten, ob vzlogger läuft und starte ihn zur Not nach. Das läuft jetzt stabil nach reboot. S0 ist auch stabil, arbeite mit pulldown, der S0-Ausgang des Zähler macht dann einen positiven Impuls, bei pullup gab es wirre Impulse.

J-A-U, 2023-02-17:
Welche GPIO nutzt ihr den jeweils?

whymanda, 2023-02-17:
17 und 27

Whitytom, 2023-02-17:
2 und 3

Whitytom, 2023-02-18:
Ich hab eine manuelle Installation von Middleware und VZlogger mit dem aktuellen Raspberry Pi OS light und den install.sh Scripten gemacht, jetzt funktioniert es wieder!

r00t-, 2023-08-09:
probably a duplicate of #526 , which has more details.

closed 2023-08-09 by r00t-

J-A-U, 2023-08-10:
We should open one of the issues and actually solve the underlying reason. #526 is, so far, just a workaround.