volkszaehler/vzlogger #578: MeterS0::HWIF_UART::waitForImpulse - Buggy return value evaluation

beckenc, 2023-02-15:
I assume, that  Line 522 should be `ret > 1`.
https://github.com/volkszaehler/vzlogger/blob/d0d27eaa720291b94ab4aa4cf09f783683c5a727/src/protocols/MeterS0.cpp#L518-L531

r00t-, 2023-02-15:
@beckenc: what's the symptoms caused by this?

that code is eight years old ... @mbehr1
https://github.com/volkszaehler/vzlogger/commit/2b2f5a87154c05938a88b534bb0b0d0ff7b3fd3d

beckenc, 2023-02-15:
Not sure of the implications but it is obviously wrong. The else path would never be traversed.

mbehr1, 2023-02-16:
I think the code is correct as is ;-)

Avoiding a busy loop in error cases might be an issue.

There is a case for <0 (error), ==0 (timeout) and (indirectly, as remaining case the regular return at the end of the function.

beckenc, 2023-02-16:
I agree, `ret == 0` is definetly a valid scenario. But, in case of `ret == 0` (that's the `else if`) `red < 1` is also true. That's the `if` path.

In case of `ret == 0` the `if` will be traversed and **NOT** the `else if`.

mbehr1, 2023-02-16:
oh! good point. 
It should be: 
if (ret < 0) ... else if (ret == 0) ...

beckenc, 2023-02-16:
Yep... That's what my PR #579 does ;-)

closed 2023-02-25 by beckenc