volkszaehler/vzlogger #580: mqtt client stops on startup when network is not yet available

otih, 2023-02-19:
on old environments (like raspberry/banana pi) vzlogger is starting to early before network is fully available.
In some configurations vzlogger is not able to start working propperly, e.g. publish data via mqtt.

[Feb 19 23:26:44][main] vzlogger v0.8.1 based on heads/master-0-g927c263441 from Mon, 2 Jan 2023 20:06:16 +0100 started.
[Feb 19 23:26:44][mqtt] MqttClient::MqttClient called
[Feb 19 23:26:44][mqtt] using libmosquitto 2.0.11
[Feb 19 23:26:44][mqtt] mosquitto_connect failed, giving up: Network is unreachable
[Feb 19 23:26:44][mqtt] ~MqttClient called
[Feb 19 23:26:44][mqtt] mqtt client not configured. stopped.

vzlogger will not retry after stopping mqtt client.

r00t-, 2023-02-20:
as for the proposed fix in #581, see my commen there: https://github.com/volkszaehler/vzlogger/pull/581#issuecomment-1437483806

r00t-, 2023-02-20:
the error handlng code for the mqtt_connect call already tries to discern retryable and non-retryable errors:
https://github.com/volkszaehler/vzlogger/blob/master/src/mqtt.cpp#L185

we should probably just add errno==ENETUNREACH as a case to be retried, maybe some others too.
maybe we should even retry ANY case of `MOSQ_ERR_ERRNO`, as those will all be network i/o errors?

otih, 2023-02-21:
will continue working as proposed on mqtt implementation and error handling. PR #581 has been closed.

narc-Ontakac2, 2023-10-28:
Short note on the network-online target: The [systemd documentation on this](https://systemd.io/NETWORK_ONLINE/) says  that the network-online is an acceptable solution if it does not delay boot but having the source code handle this is preferable.

narc-Ontakac2, 2023-10-28:
Note that there is [issue # 2878 for mosquitto](https://github.com/eclipse/mosquitto/issues/2878) which might be resolved by having mosquitto wait for network-online. In that case vzlogger waiting for networ-online might not resolve the issue.

narc-Ontakac2, 2023-10-28:
> maybe we should even retry ANY case of `MOSQ_ERR_ERRNO`, as those will all be network i/o errors?

This sounds right. Maybe retries with a decreasing frequency,  something like from 5s to 5m.

narc-Ontakac2, 2023-10-29:
Anyway, thanks for working on this. This is as serious bug, which in my case (vzlogger -> mosquitto -> openhab) could cause complete data loss after a power outage.

Gabweb, 2023-11-19:
As a temporary workaround I delayed starting the vzlogger by editing the Systemd start script. 
Maybe this is also helpful to others until the cause is resolved. (One drawback is that service restart is also affected by delay!) 

```
[Service]
ExecStartPre=/bin/sleep 10
```

narc-Ontakac2, 2023-11-21:
Waiting for the network-online target might also do it as a workaround.

narc-Ontakac2, 2023-12-29:
Should be fixed by #617.

closed 2023-12-29 by narc-Ontakac2