volkszaehler/vzlogger #583: protocol "exec"  should drop elevated privileges instead of recompiling to allow popen as root

jscheidtmann, 2023-03-12:
Protocol "exec" denies to execute any programs, if running as root, see [MeterExec.cpp - L123 ff](https://github.com/volkszaehler/vzlogger/blob/master/src/protocols/MeterExec.cpp#L123). The only option is to recompile vzlogger to allow this using a compile time configuration setting, that allows to call external programs as root.

Instead Exec should:
* Drop elevated privileges, if executed as root, before running the external script.
* Add a `Username` or `UID` parameter that specifies which uid to execute the command as in the configuration.

Example code for dropping elevated privileges can be found in "[Secure Programming Cookbook for C and C++ by John Viega, Matt Messier](https://learning.oreilly.com/library/view/secure-programming-cookbook/0596003943/)", Recipes 1.3, 1.6, 1.7.

J-A-U, 2023-03-13:
We already recommend to solve this problem by using a different user: https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version#vzlogger_als_anderer_benutzer_ausfuehren

jscheidtmann, 2023-03-13:
Ah, hadn't spotted that yet. Then:

* vzlogger should recommend doing that instead of recommending to recompile with exec as root. 
* In addition I believe it would be a nice improvement to drop privileges for the exec.

I am willing to work on pull requests to do both. Will take some time, though.

r00t-, 2023-03-17:
i don't think this is a real issue.
the default behaviour is safe, and recompiling to change behaviour is out of scope of any security considerations.

also, the more critical improvement might be to provide the infrastructure (systemd service file, udev rules) to avoid running as root?