volkszaehler/vzlogger #584: MQTT stop working

PeterMau19, 2023-03-21:
I use vzlogger with mqtt support for a gas meter (gpio pin) and electric meter (sml protocol via serial adapter) on a raspi4 and debian 
[vzlogger.conf.pdf](https://github.com/volkszaehler/vzlogger/files/11031822/vzlogger.conf.pdf)
6.1.19-v8+ .  The system works perfect but after a certain time (usually 1..2 h)  the mqtt messages for the gas meter (chn 0)  are missing. The sml messages on chn1 and chn2 are continously working. I compiled the newest version, but without effect.
Cheers Peter

r00t-, 2023-03-21:
thanks for your report, but we need something more to work with...
for a start i would suggest that you change `"verbosity": 0,` to something higher (probably 15),
(make sure you have enough diskspace for a large logfile,)
and then check the log for around the point in time where it stops working.

PeterMau19, 2023-04-13:
Nothing in vzlogger.log ( verbose 5) but I recognized the following in mosquitto.log:
#####################################################
1681228530: mosquitto version 2.0.11 terminating
1681228530: Saving in-memory database to /var/lib/mosquitto//mosquitto.db.
1681228540: mosquitto version 2.0.11 starting
1681228540: Config loaded from /etc/mosquitto/mosquitto.conf.
1681228540: Opening ipv4 listen socket on port 1883.
1681228540: Opening ipv6 listen socket on port 1883.
1681228540: mosquitto version 2.0.11 running
1681228548: New connection from 192.168.2.64:36194 on port 1883.
1681228548: New client connected from 192.168.2.64:36194 as vzlogger_69758 (p2, c1, k30).
1681228548: Client vzlogger_69758 disconnected. 
#################################################
and 
###############################################
1681381447: New connection from 192.168.2.64:59949 on port 1883.
1681381448: Client <unknown> closed its connection.
1681381448: New connection from 192.168.2.64:37413 on port 1883.
1681381448: New client connected from 192.168.2.64:37413 as 3C115538-0C36-4528-40F8-61142CCD8DB5 (p2, c1, k60).
####################################################
Vzlogger IP = 192.168.2.64

sometimes "new client" as vzlogger sometimes only ip:port

FoxWhiskey, 2024-01-30:
> I use vzlogger with mqtt support for a gas meter (gpio pin) and electric meter (sml protocol via serial adapter) on a raspi4 and debian [...]

Same here - with the difference that MQTT is not working at all. 
Configuration:
```
Debian Bookworm 12.4
vzlogger 0.8.3 (cloudfront repository)
mosquitto 2.0.11-1.2+deb12u1
```
I also found that,
* I was no longer able to compile a Debian-package by myself. I have managed this with Bullseye a couple of months ago (see #543). However, @maxberger 's fork I used back then is no longer available.
* I failed finding the required `libsml-dev`-package (>=1.0) - but I guess it's also available from the cloudfront repository (not tried yet)
* the [vzlogger.conf - Editior](http://volkszaehler.github.io/vzlogger/) no longer supports MQTT-configuration. My own configuration file contains (a formerly working) MQTT-section, most likely provided  by the online editor (cant remember, though).

J-A-U, 2024-01-30:
Your comment is very confusing.

> Same here - with the difference that MQTT is not working at all.

When it's not running at all it's not the same.

> * I was no longer able to compile a Debian-package by myself.

Does it mean you have no running vzlogger at all?
Besides, this is very OT here!

> * I failed finding the required `libsml-dev`-package

In this case you have to build from sources. See wiki for details.

> * the [vzlogger.conf - Editior](http://volkszaehler.github.io/vzlogger/) no longer supports MQTT-configuration.

It never did.

FoxWhiskey, 2024-01-30:
Alright - my excuses!
I did not mean to confuse or capture an issue - rather support finding a potential bug... Let me try again:

> When it's not running at all it's not the same.

`vzlogger` from the cloudfront-repository does install and run. I was able to observe similar log messages as you (`Client vlogger_xxx disconnected`) in `mosquitto.log`. However, this happens shortly after startup of _vzlogger.service_ and not only after 1-2hrs runtime. From my perspective, MQTT provider does not work at all (from `vzlogger.log` I know that the meter itself is working).

> Does it mean you have no running vzlogger at all?
Besides, this is very OT here!

I used to have a working, self compiled deb-package of version (`0.3.5`) on Bullseye. I have recently switched to Bookworm and was no longer able to build a deb-package (from the master branch) , which is OT in the first place. But it also suggests that the MQTT-problem was induced somewhere between `0.3.5` and `0.8.3` (latter version from the cloudfront repository)

> In this case you have to build from sources. See wiki for details.

The information on the Wiki is a bit confusing. To be more precise, it did not help building a deb-package and I did not try to compile and install _just_ with the install-script. But this OT indeed and not to be discussed here.

>It never did.
Check the history: etc/vzlogger_generic.schema.json

Ok, should have checked prior posting! But I was convinced that I haven't come up with the MQTT-configuration section by myself :wink: But anyways..

If I can contribute any data to help resolve this issue, please let me know.

r00t-, 2024-01-30:
@FoxWhiskey:
whatever you are posting here is completely unrelated to @PeterMau19 's issue /  of this ticket.
other than that you are looking to also use mqtt.

if you find issues using the vzlogger debian package(s), we (and specifically @narc-Ontakac2 ) might be interested in those,
same for issues with using mqtt.
but we suggest you try a general support forum first, to rule out issues on your end,
and then create a new ticket for the new topic. and separate ones for installation and runtime issues.

i will close this ticket, as @PeterMau19 has not responded to questions for more information for about half a year.

closed 2024-01-30 by r00t-

narc-Ontakac2, 2024-01-31:
> * I failed finding the required `libsml-dev`-package (>=1.0) - but I guess it's also available from the cloudfront repository (not tried yet)

The libsml packages on trixie or in the Cloudfront repo are recent. The >= 1.0 is there to prevent using outdated ones, you can however edit the control file and remove it. debuild should work without it.

narc-Ontakac2, 2024-01-31:
> * I was no longer able to compile a Debian-package by myself.

As said, separate issue please.