volkszaehler/vzlogger #590: read from file, instead of device

dreamonhigh, 2023-05-01:
Hello. 

I use a few raspberrypi. one is in cellar and is connected to my heater. I also connected it to my SML Grid Counter by USB-IR.
I read it by my self written program and I could host the SML data of this rpi. So I have a link where I can wget or download latest SML data from him. 
My question. is it possible to do? and how can I import/read/download those data into vzlogger? what do I have to enter in vzlogger.conf?

my download link data is updated every second with new data. 
Link: http://ip_rpi_source/SML_counterbinary.bin"

I know /dev/ttyUSB1 is a endless serial stream. my file has to be reread every second. maybe pipe it somehow. 

Thanks for reading this.

J-A-U, 2023-05-01:
No, meter from type `sml` cannot read from file. Only devices on the machine or available per URL.

We have a meter `file` (never seen an application for it), but that seem to be only for ASCI. No SML-parser attached.

Also, there is a meter of type `exec` which can call commands or scripts. It might be a possible solution to command libsml to parse your file.

dreamonhigh, 2023-05-02:
thank you.

my friend chatgpt and I found a solution.
This works for me. might be not perfect, maybe I have to test it more. but it works.

here is how I did it:

`sudo apt install tmux  // install tmux`
`tmux new -s simulate  // start a tmux session`

`sudo socat PTY,link=/dev/pty,raw,echo=0,b9600 PTY,link=/dev/ttyUSB1,raw,echo=0,b9600` 
this create a /dev/ttyUSB1device and gets data from /dev/pty ( this is running permanent in background )
now open a new terminal inside tmux by pressing CTRL+b+c

`sudo chown vzlogger:vzlogger /dev/ttyUSB1`
without this, vzlogger has no permission to read from "new device" /dev/ttyUSB1

now we have to send data to /dev/pty that gives it to /dev/ttyUSB1. on my cellar RPI, in my case I wrote sml data in a file on /run/user/1000 (ramdisk) and created a symlink to my lighttpd html server. where I can download the file very easy from every device in local net. (might be working with netcat too)

`while true; do wget -O /tmp/SML_data.bin "http://myconnected_rpi.local/SML_ZaehlerBinary.bin" && sudo sh -c 'cat /tmp/SML_data.bin > /dev/pty'; sleep 10; done`

this endless loop is downloading my SML data from "server RPI" to a static name and place in /tmp. sends it to /dev/pty and is waiting 10 sec. for next cycle.

vzlogger works fine with this virtual created /dev/ttyUSB1 
its working till now. hope this helps other users.

J-A-U, 2023-05-02:
Interesting and creative solution.

closed 2023-05-02 by J-A-U

r00t-, 2024-02-04:
@dreamonhigh:
have you noticed that vzlogger traditionally supports streaming sml over tcp?
(that's why the sml meter has "host" and "port" parameters.)
thus the most straightforward solution would be to have your program serve the sml stream on a plain tcp socket (not file on http), then vzlogger could directly read that.