volkszaehler/vzlogger #597: ```

Li900j, 2023-07-02:
```
[~/vzlogger]$ cat etc/vzlogger.conf.direction_from_status
{
    "verbosity": 15, "retry": 0,
    "meters": [{
            "enabled": true, "allowskip": false,
            "protocol": "sml",
            "device": "/home/r00t/libsml-testing/EMH-ED300L_delivery.bin",
            "direction_from_status": true,
    }]
}
[~/vzlogger]$ ./src/vzlogger -c etc/vzlogger.conf.direction_from_status -f |head -n 32
[Mar 31 05:01:22][mtr0] Got 4 new readings from meter:
[Mar 31 05:01:22][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=17243675.60 ts=1617159682423
[Mar 31 05:01:22][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=17243675.60 ts=1617159682423
[Mar 31 05:01:22][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1617159682423
[Mar 31 05:01:22][mtr0] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=-126.60 ts=1617159682423
[...]
[~/vzlogger]$ sed -i s,delivery,consumption, etc/vzlogger.conf.direction_from_status
[~/vzlogger]$ ./src/vzlogger -c etc/vzlogger.conf.direction_from_status -f |head -n 32
[...]
[Mar 31 05:01:56][mtr0] Got 4 new readings from meter:
[Mar 31 05:01:56][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=17243368.30 ts=1617159716495
[Mar 31 05:01:56][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=17243368.30 ts=1617159716495
[Mar 31 05:01:56][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1617159716495
[Mar 31 05:01:56][mtr0] Reading: id=1-0:15.7.0*255/ObisIdentifier:1-0:15.7.0*255 value=137.10 ts=1617159716495
[...]
```

_Originally posted by @r00t- in https://github.com/volkszaehler/vzlogger/issues/476#issuecomment-810722587_

Li900j, 2023-07-02:
Moj905@Outlook.sa

Li900j, 2023-07-02:
Li900j@gmail.com

r00t-, 2023-07-02:
looks like a broken spambot...

closed 2023-07-02 by r00t-