volkszaehler/vzlogger #599: CMake Error: Unknown argument -ENABLE_MQTT=on

alkomm, 2023-07-11:
I followed the regular documentation (https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version) for installing vzlogger, but unfortunately, I'm getting this error immediately when I try to install MQTT:

```
88 Tests 0 Failures 0 Ignored
OK
make[1]: Leaving directory '/home/USER/vzlogger/libs/libsml/test'
~/vzlogger/libs ~/vzlogger
~/vzlogger

building and installing vzlogger
~/vzlogger/build ~/vzlogger

building vzlogger
CMake Error: Unknown argument -ENABLE_MQTT=on
CMake Error: Run 'cmake --help' for all supported options.
```

my command:

```bash
./install.sh vzlogger libjson libsml mqtt
```

i'm on raspberry pi 4, with ubuntu server.

do somebody have an idea? what im doing wrong?

J-A-U, 2023-07-11:
Please execute in shell and post the result here.
`cd ~/vzlogger`
`git log -n 1`

J-A-U, 2023-07-11:
I can't replicate the failure over here on Raspbian.

Give it a try and run `./install.sh vzlogger`. If I'm not confused MQTT will still be included.

alkomm, 2023-07-11:
okay, there is nothing in the logfile, but i checked now with the mqtt explorer, if something comes in mqtt.
the last data is in there.

Perfect!
thank you!
it was actually the normal `./install.sh` after restart it worked for me.

closed 2023-07-14 by alkomm