volkszaehler/vzlogger #605: vzlogger - Illegal instruction

sergzin, 2023-10-17:
I installed vzlogger repo and package as described in the [Readme](https://github.com/volkszaehler/vzlogger#debian-packages), but when I run it from Raspberry Pi 1 - i get the following error. 
Is  vzlogger_0.8.2_armhf.deb  built with RPI1 armv6 architecture support? 

`serg@raspberrypi(ro):~$ vzlogger
Illegal instruction

serg@raspberrypi(ro):~$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 0002
Serial		: 00000000d569cda5
Model		: Raspberry Pi Model B Rev 1`

r00t-, 2023-10-26:
this would be a bug(?) in https://github.com/volkszaehler/vzlogger/pull/588 ?

i wonder if debian should prevent the installation of the incompatible packsge. if the packageis labelled correctlt.

@narc-Ontakac2: can we / do we want to provide a build that also works on the older raspberry pi?

narc-Ontakac2, 2023-10-27:
According to [the Debian Wiki on RPi](https://wiki.debian.org/RaspberryPi#Raspberry_Pi_1_.28A.2C_B.2C_A.2B-.2C_B.2B-.2C_Zero.2C_Zero_W.29) RPi 1 should work with Debians armel architecture. We should be able to build for that without much effort. Unfortunately I am currently sick, this will take a few days.

narc-Ontakac2, 2023-10-31:
There are a vzlogger 0.8.3 and a libsml 1.1.1 now that are also packaged for the armel Debian architecture. @sergzin could you please confirm if they work on your RPi 1.

r00t-, 2023-11-03:
@narc-Ontakac2: thanks!

J-A-U, 2024-02-19:
For me armel doesn't do the trick.

```pi@pi:~ $ curl -1sLf   'https://dl.cloudsmith.io/public/volkszaehler/volkszaehler-org-project/setup.deb.sh'   | sudo -E distro=debian codename=bookworm arch=armel bash
Executing the  setup script for the 'volkszaehler/volkszaehler-org-project' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Checking for apt signed-by key support ...
   OK: Importing 'volkszaehler/volkszaehler-org-project' repository GPG keys ...
   OK: Checking if upstream install config is OK ...
   OK: Installing 'volkszaehler/volkszaehler-org-project' repository via apt ...
   OK: Updating apt repository metadata cache ...
   OK: The repository has been installed successfully - You're ready to rock!

pi@pi:~ $ sudo apt-get install vzlogger
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libmicrohttpd12 libmosquitto1 libsml1
Suggested packages:
  libsml-utils
The following NEW packages will be installed:
  libmicrohttpd12 libmosquitto1 libsml1 vzlogger
0 upgraded, 4 newly installed, 0 to remove and 6 not upgraded.
Need to get 787 kB of archives.
After this operation, 6,093 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian bookworm/main armhf libmicrohttpd12 armhf 0.9.75-6 [107 kB]
Get:3 https://dl.cloudsmith.io/public/volkszaehler/volkszaehler-org-project/deb/debian bookworm/main armhf libsml1 armhf 1.1.2 [18.3 kB]
Get:2 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf libmosquitto1 armhf 2.0.11-1.2+deb12u1 [78.3 kB]
Get:4 https://dl.cloudsmith.io/public/volkszaehler/volkszaehler-org-project/deb/debian bookworm/main armhf vzlogger armhf 0.8.3 [583 kB]
Fetched 787 kB in 2s (401 kB/s)
Selecting previously unselected package libmicrohttpd12:armhf.
(Reading database ... 86720 files and directories currently installed.)
Preparing to unpack .../libmicrohttpd12_0.9.75-6_armhf.deb ...
Unpacking libmicrohttpd12:armhf (0.9.75-6) ...
Selecting previously unselected package libmosquitto1:armhf.
Preparing to unpack .../libmosquitto1_2.0.11-1.2+deb12u1_armhf.deb ...
Unpacking libmosquitto1:armhf (2.0.11-1.2+deb12u1) ...
Selecting previously unselected package libsml1:armhf.
Preparing to unpack .../libsml1_1.1.2_armhf.deb ...
Unpacking libsml1:armhf (1.1.2) ...
Selecting previously unselected package vzlogger.
Preparing to unpack .../vzlogger_0.8.3_armhf.deb ...
Unpacking vzlogger (0.8.3) ...
Setting up libsml1:armhf (1.1.2) ...
Setting up libmosquitto1:armhf (2.0.11-1.2+deb12u1) ...
Setting up libmicrohttpd12:armhf (0.9.75-6) ...
Setting up vzlogger (0.8.3) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u4) ...
pi@pi:~ $ vzlogger
Illegal instruction
```

Might be interesting for you to know how Raspbian identifies himself:
```
pi@pi:~ $ curl -1sLf \
  'https://dl.cloudsmith.io/public/volkszaehler/volkszaehler-org-project/setup.deb.sh' \
  | sudo -E bash
Executing the  setup script for the 'volkszaehler/volkszaehler-org-project' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=raspbian  version=12  codename=bookworm  arch=armv6l
 >>>>:
[...]
```

narc-Ontakac2, 2024-02-19:
Please try 
```
sudo apt-get remove libsml1 vzlogger
sudo apt-get install libsml1:armel vzlogger:armel
```
For some reason apt-get installs armhf.

If that fails installing the downloaded debs might do.

J-A-U, 2024-02-20:
> sudo apt-get install libsml1:armel vzlogger:armel

```pi@pi:~ $ sudo apt-get install libsml1:armel vzlogger:armel
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libsml1:armel
E: Unable to locate package vzlogger:armel
```

> If that fails installing the downloaded debs might do.

```pi@pi:~ $ sudo apt install ~/libsml1_1.1.2_armel.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libsml1:armel' instead of '/home/pi/libsml1_1.1.2_armel.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libsml1:armel : Depends: libc6:armel (>= 2.15) but it is not installable
                 Depends: libuuid1:armel (>= 2.16) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

oder:

```pi@pi:~ $ sudo dpkg --install ~/libsml1_1.1.2_armel.deb
dpkg: error processing archive /home/server/libsml1_1.1.2_armel.deb (--install):
 package architecture (armel) does not match system (armhf)
Errors were encountered while processing:
 /home/server/libsml1_1.1.2_armel.deb
```

narc-Ontakac2, 2024-02-20:
Looks like your Raspian considers itself as armhf. To confirm please run 
```
dpkg --print-architecture
```

J-A-U, 2024-02-20:
```
pi@pi:~ $ dpkg --print-architecture
armhf
```

narc-Ontakac2, 2024-02-21:
Ouch, this is bad news. It means there is no way to build an RPi 1 specific package. I am wondering how Raspian does it. Do they have different repostories for the differen RPis?

narc-Ontakac2, 2024-02-21:
OK, I think I'm getting closer. I took a look at the Raspian mosquitto 2.0.18 rules file, which includes buildflags.mk.
```
$ apt-file search buildflags.mk
dpkg-dev: /usr/share/dpkg/buildflags.mk
```
Looking at the Raspbian repo I found that they don't have a dpkg-dev package (the -el packages are for emacs). 

What does `apt-file search buildflags.mk` give on  the RPi 1?

J-A-U, 2024-02-21:
Hm?!
```
pi@pi:~ $ apt-file search buildflags.mk
-bash: apt-file: command not found
```

Maybe helpful:
```
pi@pi:~ $ dpkg-buildflags --export
export ASFLAGS=""
export CFLAGS="-g -O2 -ffile-prefix-map=/etc/dpkg=. -fstack-protector-strong -Wformat -Werror=format-security"
export CPPFLAGS="-Wdate-time -D_FORTIFY_SOURCE=2"
export CXXFLAGS="-g -O2 -ffile-prefix-map=/etc/dpkg=. -fstack-protector-strong -Wformat -Werror=format-security"
export DFLAGS="-frelease"
export FCFLAGS="-g -O2 -ffile-prefix-map=/etc/dpkg=. -fstack-protector-strong"
export FFLAGS="-g -O2 -ffile-prefix-map=/etc/dpkg=. -fstack-protector-strong"
export GCJFLAGS="-g -O2 -ffile-prefix-map=/etc/dpkg=. -fstack-protector-strong"
export LDFLAGS="-Wl,-z,relro"
export OBJCFLAGS="-g -O2 -ffile-prefix-map=/etc/dpkg=. -fstack-protector-strong -Wformat -Werror=format-security"
export OBJCXXFLAGS="-g -O2 -ffile-prefix-map=/etc/dpkg=. -fstack-protector-strong -Wformat -Werror=format-security"
```

narc-Ontakac2, 2024-02-27:
This will take some time.

narc-Ontakac2, 2024-03-09:
For simplicity I used libsml to do a raspbian build. Please test if at least `sml_server -h` works.

[image]
[image]

J-A-U, 2024-03-10:
Yes, this version works:
```pi@pi:~ $ sml_server -h
usage: sml_server [-h] [-s] [-v] device
device - serial device of connected power meter e.g. /dev/cu.usbserial, or - for stdin
-h - help
-s - process only one OBIS data stream (single)
-v - verbose
```

narc-Ontakac2, 2024-03-10:
I have started adapting the Release action to also do RPi builds. I'll do libsml first. Unfortunately the release build currently fails because the build for trixie fails. Trixie is testing, so this hopefully will go away by itself within a few days.

r00t-, 2024-03-11:
@narc-Ontakac2: thanks for keeping up the work on this!

narc-Ontakac2, 2024-03-29:
@J-A-U I got the libsml trixie build to work again. So cloudsmith has now libsml Raspbian packages for bullseye and bookworm. You need to replace debian with raspbian in the repository URL in addition to the distro to use that. Could you please test that.

If that works I'll apply it to vzlogger.

J-A-U, 2024-03-29:
> So cloudsmith has now libsml Raspbian packages for bullseye and bookworm. You need to replace debian with raspbian in the repository URL in addition to the distro to use that.

To assign the new build to raspbian means the cloudsmith script should work properly. And it does!
So does libsml1. Great work!

I don't remember clearly. Do we need a arm64-variant as well for the script 'https://dl.cloudsmith.io/public/volkszaehler/volkszaehler-org-project/setup.deb.sh' as well? The debian arm64 build was running, but if i'm not mistaken i had to specify manually.

narc-Ontakac2, 2024-03-29:
It looks like the Raspian team is working on an [arm64 Raspian](https://downloads.raspberrypi.org/raspios_arm64/images/). Not sure about the status.

What is that setup.deb.sh script? I didn't know about that.

J-A-U, 2024-03-29:
> It looks like the Raspian team is working on an [arm64 Raspian](https://downloads.raspberrypi.org/raspios_arm64/images/). Not sure about the status.

I tried it several months ago. Vzlogger in debian arm64 build is working on these. But only newer hardware supports 64bit.

> What is that setup.deb.sh script? I didn't know about that.

By Cloudsmith recommended installation method. https://cloudsmith.io/~volkszaehler/repos/volkszaehler-org-project/setup/#formats-deb
We link to this page from ./README.md.

narc-Ontakac2, 2024-04-02:
@J-A-U There is a Raspbian vzlogger 0.8.4 in the cloudsmith repo now. Please test.

r00t-, 2024-05-08:
should be maybe add code somewhere that detects this situation and warns the user, i.d. in a preinst script?
(@narc-Ontakac2)

J-A-U, 2024-05-08:
Warn for what?

r00t-, 2024-05-08:
have the debian-armhf package detect when it's installed on a raspberry pi 1 and warn the user that it won't work.

narc-Ontakac2, 2024-05-09:
Doable, but a bit harder to do it reliably. We need to detect a certain CPU architecture if the preinstall script is build for Debian. Any hit, how and which CPU architecture we need to detect?

r00t-, 2024-05-09:
it seems to be as simple as:
```
# grep Model /proc/cpuinfo
Model           : Raspberry Pi Model B Rev 2
```
i only wonder if the raspberry pi world has a _recommended_ method to detect the model.
i would have expected that `raspi-config` is complemented by `raspi-info` or such, but that does not seem to be the case.