volkszaehler/vzlogger #609: Get live data from HTTP server with active aggregation

shikousa, 2023-11-04:
The HTTP server seems not to provide realtime readings, but only the aggregated values.
 
This delivers exactly the same, but should be different, as my aggregation time is 600s.
http://localhost/middleware.php/data.json?options=raw&from=now&uuid[]=1111-2222-3333
http://localhost:8081/1111-2222-3333

Is there a way to get the live data over HTTP? The push server is working, as I can see the live data in the middleware.

r00t-, 2023-11-04:
the server serves dsta from `localbuffer`:
https://github.com/volkszaehler/vzlogger/blob/master/src/local.cpp#L53

which is populated with data from just after aggregation:
https://github.com/volkszaehler/vzlogger/blob/df03cf606fba77cd8dc8aa90586df7284dadad14/src/threads.cpp#L166

i'm not sure how much changes would be needed to serve non-aggregated data.
(besides obvious things like requiring a larger buffer.)

shikousa, 2023-11-05:
During aggregation it seems that both Push Server and MQTT Data is updated. 
Is MQTT really updated during aggregation, as there is another update after aggregation?
https://github.com/volkszaehler/vzlogger/blob/df03cf606fba77cd8dc8aa90586df7284dadad14/src/threads.cpp#L136-L147

Is it possible to add this code section also during aggregation?
https://github.com/volkszaehler/vzlogger/blob/df03cf606fba77cd8dc8aa90586df7284dadad14/src/threads.cpp#L163-L168

r00t-, 2023-11-09:
> Is it possible to add this code section also during aggregation?

why don't you just try it, worst case it will crash...

> Is MQTT really updated during aggregation, as there is another update after aggregation?

i'm not so familiar with the MQTT integration, but i think it publishes BOTH the raw and aggregated values.

r00t-, 2024-12-20:
@indus:
if the holidays have me bored enough to attempt implementing this, would you be available to test? [can you build a branch from source or are you using binaries?]

indus, 2024-12-20:
I used to use the binaries on an arm SoC and switched to a docker container recently with a transition to amd64 hardware. I probably could test between the holidays.

indus, 2024-12-20:
@r00t- I used to use the binaries on an arm SoC and switched to a docker container recently with a transition to amd64 hardware. I probably could test ~~between the holidays~~ between Christmas and New Year’s Day (“zwischen den Jahren”)

I guess it should be easy to let the Dockerfile check-out a branch.

indus, 2024-12-28:
I've now switched to using a custom middleware and the [push option](https://wiki.volkszaehler.org/software/controller/vzlogger/vzlogger_conf_parameter#push), so there's no longer any need for this feature on my side.

For anyone who might find it useful, here's how my Deno server looks:

```
import { Hono } from 'npm:hono';

const app = new Hono();

app.post("/", async (c) => {
    try {
        const body = await c.req.json();

        if (body.data && Array.isArray(body.data)) {
            for (const item of body.data) {
                if (item.uuid === UUID) {
                    // business logic
                }
            }

            return c.json(body, 200);
        } else {
            return c.text("Invalid data format", 400);
        }
    } catch (error) {
        return c.text("Internal Server Error", 500);
    }
});

Deno.serve({ port: SERVER_PORT }, app.fetch);

```

r00t-, 2025-01-09:
(i find it irritating to see a long listing of mostly boilerplate code in an obscure language that this project does not use posted here.)

indus, 2025-01-09:
> (i find it irritating to see a long listing of mostly boilerplate code in an obscure language that this project does not use posted here.)

sorry - i deleted it (my idea was that it may have helped others to build a workaround in a similar situation)

r00t-, 2025-01-10:
sorry, i didn't mean you have to delete it.
but it's so overly specific to that language/library you used and contains so little code specific to vzlogger.
now if it was at least very short and concise, it could serve as an advertisement for using that language, but it's not even that.

plus, the reference to the [push option](https://wiki.volkszaehler.org/software/controller/vzlogger/vzlogger_conf_parameter#push) is very valid, as it provides a way to extract the unaggregated data.
your providing of some code is also correct in the sense that there is no specific documentation of the "push" data format (as far as i checked).
(just that that was only three lines of the twenty lines of code you posted, and not specificlly marked as such.)

actually, now that i read the code for "push" in vzlogger, it has a problem similar to mqtt, => https://github.com/volkszaehler/vzlogger/issues/610
(which is also why it can (intentionally) bypass aggregation.)

indus, 2025-01-11:
No problem - and I understand your objection. I think I found a compromise for my deleted post:

> I've now switched to using a custom middleware and the push option, so there's no longer any need for this feature on my side. 

<details>
<summary>Deno Code Example</summary>

```JS
import { Hono } from 'npm:hono';

const app = new Hono();

app.post("/", async (c) => {
    const body = await c.req.json();

    for (const channel of body.data) {
        if (channel.uuid === UUID) {
            // business logic
        }
    }

    // respond with a JSON or the push sever throws an error
    return c.json(body, 200);
})

Deno.serve({ port: SERVER_PORT }, app.fetch);
```
</details>

r00t-, 2025-01-16:
i still think your example would be much more useful if it would show the format of the data, i.e. with a debug printf,
and it's surely excessive to echo back the whole request?

indus, 2025-01-16:
> it's surely excessive to echo back the whole request?

I'm really not good at PHP (nor C++), but I had a look at the official code, and it seems to me it is doing something very similar.

To my understanding it recreates the channel data here (with the difference that it may be `transformed` by some `interpreter`):
https://github.com/volkszaehler/volkszaehler.org/blob/9d83e1d6c6ae85f7dd4445b5f6e79586124872e3/lib/Server/MiddlewareAdapter.php#L186-L189

and sends it back here:
https://github.com/volkszaehler/volkszaehler.org/blob/9d83e1d6c6ae85f7dd4445b5f6e79586124872e3/lib/Server/HttpReceiver.php#L71

I don't know why the official code does so, but sending the whole body back comes close and shoudn't be more excessive. What I really don't understand is what the push server does with the information?

I think the relevant is this check here:
https://github.com/volkszaehler/vzlogger/blob/ae0372d60114cfa01924bcc9dfda7a26be317c4c/src/PushData.cpp#L174-L191
But its purpose? 😶 I have no idea!

r00t-, 2025-01-21:
vzlogger only stores the response-body to dump it in case of an error:
```
		} else if (http_code != 200) {
			print(log_alert, "CURL Error from url %s: %d %s", "push", middleware.c_str(), http_code,
				  response.data);
			toRet = false;
		}
	}

	if (response.data)
		free(response.data);
```

indus, 2025-01-22:
ok - but it also expects some data if the http_code is in fact 200, otherwise it throws an error itself. and the middleware seems to always send it not only in case of an error.

Nevertheless I've changed my example to respond with an empty object `{}`.