volkszaehler/vzlogger #615: measurements discarded due to invalid timestamp evaluation in version 0.83

KanteTaete, 2023-12-21:
I compiled version 0.83 myself from commit `df03cf606fba77cd8dc8aa90586df7284dadad14` branch: `master` with option: `./install.sh vzlogger` and adaption of `Meter.cpp` ( changed value of `METER_DETAIL(exec, Exec, "Parse program output", 32, false` to 128).

I observed in Grafana that the graph lines are very often broken lines, i.e. many measured values are not logged to the database.
This behavior is different to version 0.80 where it works perfectly with the same config and adaption.

Please refer a part of the log (verbosity level 10):
Dec 20 22:27:24][exec] MeterExec::read: Reading line: '1703107640.0769758: Warnungs_Flags = 0'
[Dec 20 22:27:24][exec] MeterExec::read: string: Warnungs_Flags, value: 0.000000, timestamp: 1703107640.076976
[Dec 20 22:27:24][exec] MeterExec::read: Closing process 'python3 /...'
[Dec 20 22:27:24][mtr0] meter returned readings with a timestamp before 1990, IGNORING.
[Dec 20 22:27:24][mtr0] most likely your meter is misconfigured,
[Dec 20 22:27:24][mtr0] for sml meters, set "use_local_time": true in vzlogger.conf (meter section),
[Dec 20 22:27:24][mtr0] to override this check, set "i_have_a_time_machine": true in vzlogger.conf.
[Dec 20 22:27:24][chn1] ==> number of tuples: 0
[Dec 20 22:27:24][chn0] Buffer has 0 items
[Dec 20 22:27:24][chn1] JSON request body is null. Nothing to send now.
[Dec 20 22:27:24][chn0] Nothing to send to InfluxDB api
[Dec 20 22:27:24][chn2] Buffer has 0 items

I can see is that 2 readings (2 channels) are evaluated with timestamp 0, but for all other channels the timestamp is different to 0 and correct.
I think, only the reading of these 2 channels may be discarded, but for the others (10 chn in total) the data should be correctly interpreted and sent to the database. But as you can see, for chn 0-2 no items are detected - which is the same for chn 3-10.

Furthermore all further readings (10s cycle time) are discarded up to timestamp Dec 20 22:27:49 (~25 seconds later). Then all readings are evaluated correctly again.

I hope this information is enough for analysis. If not, I can anonymize my log and paste the whole log & config.

r00t-, 2023-12-24:
the log message you are seeing was added in https://github.com/volkszaehler/vzlogger/pull/491 :
https://github.com/r00t-/vzlogger/blame/f613480d33339537ff3270f3f9ea058b6504ee48/src/threads.cpp#L83-L87

but this can not be the cause of the issue,
this only triggers if the parsed timestamp is far in the past, which would break your graphs either way,

in the example you gave, the timestamp of '1703107640.076976' looks valid,
but that is the raw string before parsing.

can you either:
- add log output of the parsed timestamp value
- provide your vzlogger.conf and an example output of your python program that you consider valid but that reproduces the issue?

MeterExec has not been changed in four years...

but i see that MeterExec considers the line valid if at least ONE field was parsed:
https://github.com/r00t-/vzlogger/blame/f613480d33339537ff3270f3f9ea058b6504ee48/src/protocols/MeterExec.cpp#L195
that is possibly a bug.
it seems possible that 'sscanf()' failed to parse the timestamp, and it's default value of '-1' was used,
which would trigger the timestamp in the past warning,

(one possible reason might be that you have a german locale set, and it expects a '.' as decimal separator - just a wikd guess,)

KanteTaete, 2023-12-25:
Thanks for the reply and initial analysis!
Please find below my config, the output of the python script and one more log with verbosity 15 + my analysis of the log.

**vzlogger.conf:**
[Issue615_vzlogger_conf.txt](https://github.com/volkszaehler/vzlogger/files/13765701/Issue615_vzlogger_conf.txt)

**Python output:**
python3 /home/pi/svn_wk/trunk/python/Read_ABB_B23_mbrtu.py
1703500600.1748216: Br10_Wirkleistung_gesamt = -234597
1703500600.1748216: Br10_Ausgang_1 = 1
1703500600.1748216: WP_P_Import = 766315
1703500600.1748216: WP_Wirkleistung_gesamt = 5836
1703500600.1748216: WP_Ausgang_1 = 0
1703500600.1748216: WP_Ausgang_2 = 0
0: Br8_P_Import = 0
1703500600.1748216: Br8_Strom_N = 4294967295
1703500600.1748216: Br8_Wirkleistung_gesamt = 0
1703500600.1748216: Br8_Eingang_3_aktuell = 65535

I removed the lines which's format doesn't look different to the others and are not used by any vzlogger channel.
I kept the lines containing "Br8_P_Import" and "Strom_N" although not used by any vzlogger channel as these contain an "invalid" timestamp resp. a very big value.
I hope that the line "Strom_N" is just not regarded by vzlogger as the identifier is not used by any channel and that the big value doesn't cause any overflow, etc. At least in version 0.80 this caused no problem...

**Log-Output:**
[Issue615_vzlogger_log.txt](https://github.com/volkszaehler/vzlogger/files/13766124/Issue615_vzlogger_log.txt)

I removed many entries (replaced by ...) where e.g. CURL commands were logged or valid timestamps were present for readings to point to the issue. I kept the initial successful execution of the exec-meter and then one later where it fails.

**Observation1:**
When 1 reading contains an invalid timestamp, then all readings are discarded. E.g.
[Dec 25 11:35:58][exec] MeterExec::read: Reading line: '0: WP_P_Import = 0'
[Dec 25 11:35:58][exec] MeterExec::read: string: WP_P_Import, value: 0.000000, timestamp: 0.000000
--> 
[Dec 25 11:35:58][mtr0] meter returned readings with a timestamp before 1990, IGNORING.
...Nothing to send to InfluxDB api

**Expectation1:**
Only the channel with the invalid timestamp shall be discarded, all other channels' data shall be transferred to the database.

**Observation2:**
exec-meter is executed at timestamps [Dec 25 11:35:27], [Dec 25 11:35:41], [Dec 25 11:35:54], [Dec 25 11:36:08]
- the interval time (10s) is added to the execution time (~4s). So no exact 10s intervals are possible (--> needs to be regarded during configuration), not relevant for the current issue.
- the 3rd execution at 11:35:54 "fails".

KanteTaete, 2023-12-25:
As far as I see, #491 was implemented in 2022 and is contained in v0.8.3. My version 0.8.0 is based on a commit in 2021.
So likely discarding readings with an "invalid" timestamp is the difference between the 2 versions which "causes" the issue on my end.

> this only triggers if the parsed timestamp is far in the past, which would break your graphs either way,
Yes, I agree.
But I haven't observed this in the past with v0.8.0, likely as I have never checked for the entries at timestamp 0.
I should find those values in the sql db, shouldn't I?
```
MariaDB [volkszaehler]> select * from data where timestamp = 0;
Empty set (2.749 sec)
```
Is my sql statement correct? - If so, then either 
- version v0.8.0 doesn't write any data with timestamp "0" to the db (python script never returns timestamp 0) or 
- the channels logged into sql (not all are logged there, many only to influxdb) never got an invalid timestamp (returned from the python script) but others do.

In influxdb I suppose to get an error when trying to write the data as it's out of the measurement's policy. I think I saw those in the past already in some logfile but never figured out the reason... :-)

Was it intended that if one reading of a meter has an invalid timestamp, all other readings of this meter are discarded?
I think it was better to only discard the respective channel.

r00t-, 2023-12-25:
great that you found the error outside of our code :grin:.

i guess the reason to ignore all readings is that the data structure they are stored in does not really support manipulations,
that is already kind of logged as https://github.com/volkszaehler/vzlogger/pull/585 .

(it's also a weird design that each reading has it's own timestamp, this doesn't make any sense for most meters, see https://github.com/volkszaehler/vzlogger/pull/455 . )

i suggest you simply fix your script to not generate entries with invalid timestamps.

feel free to file feature-requests for any undesired behaviour.

r00t-, 2023-12-26:
btw, you could also simply set `i_have_a_time_machine` to not discard the data. i added that option for a reason :grin:.

i would guess that the script never generated invalid timestamps for any readings you actually logged, that would explain why you have none in your database.

(we might apply the check only to values actually logged, but i don't think it's our resposibility to cater to broken meters. if the input is garbage, it should be fixed at the source.)

KanteTaete, 2023-12-27:
> (we might apply the check only to values actually logged, but i don't think it's our resposibility to cater to broken meters. if the input is garbage, it should be fixed at the source.)
Well: Yes and No...
Of course, every data source should provide valid data.
But being fault-tolerant towards invalid data marks good software. :-D
To be fault-tolerant was esp. helpful for "open inputs" as the exec meter.

I just wonder why one of the channels can get a timestamp "0" as the data is generated in a for loop using the same variable as timestamp for all channels (datapoints). So either for the channel no data should be present or "all" (incl. timestamp):
```
for meter_datapoint in current_meter.meter_data:
    if int(y[1]) == meter_datapoint.start_reg:
        #write data to meter datapoint
        meter_datapoint.data = int(y[2])
        meter_datapoint.data_timestamp = timestamp
        if debugging:
            print ("\t\tRegister:", meter_datapoint.start_reg, "Data: ", meter_datapoint.data)
        #break for loop as correct datapoint found
        break
    else:
       ...
```

r00t-, 2023-12-27:
as i said:
* the current implementation discards all values because it's impractical to remove single values from an array
* feel free to submit a feature request. (or better, submit a PR for https://github.com/volkszaehler/vzlogger/issues/490 ;) )

sorry, but debugging your python code is really out of scope.
especially as it seems to be unpublished private code?
else i'd have suggested to open a ticket with that project instead of here.

closed 2024-01-30 by r00t-