volkszaehler/vzlogger #621: Message type 900 & ENABLE_MQTT Error

Piranha1511, 2024-01-20:
Hi,

I followed following instruction: https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version

As soon as I use the command "./install.sh vzlogger libjson libsml mqtt" I get the following errors:

1.  libsml: error: message type 0900 not yet implemented
2. CMake Error: Unknown argument -ENABLE_MQTT=on

I've tried it with Raspbian Lite and also with DietPi. Both with the same results. 

System was updated before with apt update and apt upgrade.

Piranha1511, 2024-01-20:
[image]

narc-Ontakac2, 2024-01-25:
"error: message type 0900 not yet implemented" is a successful test, that checks for an error message. I have adapted the title.

r00t-, 2024-01-30:
this is caused by a trivlal syntax error in install.sh,
it seems  that nobody ever tried to use the mqtt option before.

i would personally vote to remove install.sh,
it's some attempt to make source installation easier,
in a nonstandard ad-hoc way, and it's failing, as demonstrated here.
we have (deb) packaging now,
the issues that install.sh tries to solve (installation of dependencies) should have better and more standardized soltions.
i.e. we have a redundant one with the dockerfile (which might aswell invoke install.sh, but chooses not to..

fm-maniac, 2024-01-30:
I used install.sh with mqtt option several times to update vzlogger and it worked. Lately to update to 0.8.3.

r00t-, 2024-01-30:
left a comment at the commit this was introduced at:
https://github.com/volkszaehler/vzlogger/commit/27eb8d1566ec493d9cf64d2b53676c846ebb6e35#r138085452
> i wonder where the `-<var>=<value>` instead of `-D<var>=<value>` is from,
> at least my version of cmake does not have it, i cant find a mention of it in any documentation,
> at https://github.com/volkszaehler/vzlogger/issues/621 a user reports it doesn't work for him, while others report it's working.

r00t-, 2024-02-04:
@fm-maniac:
i'd like to understand how that code worked for you...
would appreciate if you could reproduce that and give more details.
(output of `sh -x install.sh mqtt`, without the fix that i merged now.)

closed 2024-02-04 by r00t-