volkszaehler/vzlogger #624: remove install.sh (or maintain it better)

r00t-, 2024-01-30:
split from https://github.com/volkszaehler/vzlogger/issues/621
> this is caused by a trivlal syntax error in install.sh,
> it seems that nobody ever tried to use the mqtt option before.
>
>i would personally vote to remove install.sh,
> it's some attempt to make source installation easier,
> in a nonstandard ad-hoc way, and it's failing, as demonstrated here.
> we have (deb) packaging now,
> the issues that install.sh tries to solve (installation of dependencies) should have better and more standardized soltions.
> i.e. we have a redundant one with the dockerfile (which might aswell invoke install.sh, but chooses not to..

r00t-, 2024-01-30:
further issues in a quick attempt to test install.sh:
* it invoked `ldconfig`, failing because that's not on `$PATH` for non-root users,
* then tries to invoke sudo, assuming it's running as a regular user
* ir assumes debian-like package names?

J-A-U, 2024-02-04:
> we have (deb) packaging now,

Install.sh would have been purged from Wiki already if the deb-package had been working for me.
https://lists.volkszaehler.org/pipermail/volkszaehler-users/2023-May/017042.html

r00t-, 2024-02-04:
@J-A-U:
> Install.sh would have been purged from wiki already if the deb-package had been working for me.
https://lists.volkszaehler.org/pipermail/volkszaehler-users/2023-May/017042.html

slightly off the topic,
but i guess that's the bug that was reported and fixed at https://github.com/volkszaehler/vzlogger/issues/605
i'mafraid i stopped reading the `-users` list long ago, just too much noise.

narc-Ontakac2, 2024-02-18:
@J-A-U Could you please confirm if it is actually fixed. 

I have added the armel architecture as a fix that should work on RPi  1. However I do not have the hardware to check it.

narc-Ontakac2, 2024-05-11:
If install.sh stays (and OMS is a strong argument to keep it) the vzlogger.service file should be aligned with the Debian version. Currently the service installed by install.sh runs as root. There are also other minor shortcomings.