volkszaehler/vzlogger #642: VZLOGGER V0.8.3 without "oms" protocol

escribana, 2024-03-30:
Compiling by hand (following [https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version](url)) VZLOGGER V0.8.3 does not include the "oms" protocol.
"/usr/local/bin/vzlogger -h" gives
....
following protocol types are supported:
	file        	Read from file or fifo
	exec        	Parse program output
	random      	Generate random values with a random walk
	fluksov2    	Read from Flukso's onboard SPI fifo
	s0          	S0-meter directly connected to RS232
	d0          	DLMS/IEC 62056-21 plaintext protocol
	sml         	Smart Message Language as used by EDL-21, eHz and SyM²
	w1therm     	W1-therm / 1wire temperature devices
....

Falling back to V0.8.1 "oms" is present:
...
following protocol types are supported:
	file        	Read from file or fifo
	exec        	Parse program output
	random      	Generate random values with a random walk
	fluksov2    	Read from Flukso's onboard SPI fifo
	s0          	S0-meter directly connected to RS232
	d0          	DLMS/IEC 62056-21 plaintext protocol
	sml         	Smart Message Language as used by EDL-21, eHz and SyM²
	w1therm     	W1-therm / 1wire temperature devices
	**oms         	OMS (M-BUS) protocol based devices**
...

J-A-U, 2024-03-31:
Is the protocol actually missing (not working) or just not shown in the list.
As far as i understand the code, OMS is still activated by default.

escribana, 2024-03-31:
It's definitively not working! I've got a status 78 ("ExecStart=/usr/local/bin/vzlogger -c /etc/vzlogger.conf (code=exited, status=78)") with ""protocol": "oms" in vzlogger.conf.
With the very same config file version 0.8.1 runs without objection.

r00t-, 2024-04-01:
https://github.com/volkszaehler/vzlogger/blob/ec723ec71174566e93c92d6ab79bc9abfba44e6e/src/Meter.cpp#L71-L72

https://github.com/volkszaehler/vzlogger/blob/ec723ec71174566e93c92d6ab79bc9abfba44e6e/CMakeLists.txt#L56-L58

https://github.com/volkszaehler/vzlogger/blob/ec723ec71174566e93c92d6ab79bc9abfba44e6e/CMakeLists.txt#L56-L58

none of this code has changed since 2015.

how exactly did you build vzlogger?
what are the values of ENABLE_OMS and OMS_SUPPORT in your CMakeCache?
did you cleanup between switching versions? (`make clean-all` or such)

J-A-U, 2024-04-02:
Oh my... 😅

[https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version?s[]=building&s[]=oms#build_von_hand](https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version?s[]=building&s[]=oms#build_von_hand)
> Building OMS
> OMS erfordert openssl und libmbus >=0.8.0 z.B. von https://github.com/rscada/libmbus.git
> ```
> git clone https://github.com/rscada/libmbus.git
> cd libmbus
> ./build.sh
> sudo make install
> ```

But, not enough time to confirm right now.

escribana, 2024-04-02:
> Oh my... 😅
> 
> > Building OMS
> > OMS erfordert openssl und libmbus >=0.8.0 z.B. von https://github.com/rscada/libmbus.git
> > ```
> > git clone https://github.com/rscada/libmbus.git
> > cd libmbus
> > ./build.sh
> > sudo make install
> > ```

escribana, 2024-04-02:
Sorry about completing and reopening! 
I will try [https://github.com/rscada/libmbus.git](https://github.com/rscada/libmbus.git) as soon as possible.

closed 2024-04-02 by escribana

r00t-, 2024-04-02:
i didn't previously go as far as trying to install libmbus on my system,
but that has always been a dependency, and is checked by the (unchanged) code:
https://github.com/volkszaehler/vzlogger/blob/ec723ec71174566e93c92d6ab79bc9abfba44e6e/CMakeLists.txt#L98-L99
so if you can build "0.8.1" with OMS, then "0.8.3" should find it aswell.
(it works fine for my with libmbus installed.)

or could it be that you used a binary of the older version and did not actually build it yourself?
that was included in my question of "how did you build vzlogger", both the version without AND the version with oms support...

J-A-U, 2024-04-02:
> (it works fine for my with libmbus installed.)

Here as well.
When i purge install.sh from the wiki, it might be less confusing.

escribana, 2024-04-03:
Ok, V0.8.3 is working now! I made "vzlogger" 0.8.3 once again and voilá: "oms" is present. I don't know why.
Sorry for the noise.

narc-Ontakac2, 2024-04-03:
> Same problem with the new build from cloudsmith

Unfortunately libmbus is not packaged. I'll keep that in mind, but unless someone insists it is important this might never happen.

narc-Ontakac2, 2024-06-29:
The Debian/Raspian packages for vzlogger 0.8.7 are build with OMS support. @escribana In case you are using vzlogger with OMS support now I would appreciate if you could give it a try.