volkszaehler/vzlogger #645: Version information is inconsistent

J-A-U, 2024-04-04:
Vzlogger doesn't identify itself in the same way, even tho the builds should be from similar sourcecode.

Binary from cloudsmith:
```
pi@pi:~ $ vzlogger -V
0.8.4
 based on git version: GIT-NOTFOUND
 last commit date:
```

Local build:
```
pi@pi:~ $ ./vzlogger/build/src/vzlogger -V
0.8.3
 based on git version: tags/v0.8.4-0-ga56f6281fc
 last commit date: Tue, 2 Apr 2024 06:38:32 +0200
```

r00t-, 2024-04-04:
it's normally intended that public releases have just  a version number, while git/developer builds have the git commit details.

the biggest immediate issue is see is that the git info should probably be omitted altogether rather than having "GIT-NOTFOUND" or an empty string.

if i got this right, the debian build process downloads and extracts a tarball, so the git info is simply not available, which is intentional.

J-A-U, 2024-04-04:
For the public release: Got it.

It's still strange the local build identifies as 0.8.3.