volkszaehler/vzlogger #652: MQTT not publishing any messages

Coxcopi, 2024-05-05:
MQTT does not seem to be working properly. Using the repository and building with `./install.sh vzlogger libjson libsml mqtt` and then setting up MQTT in the `vzlogger.cfg` file, MQTT connects to the server properly, but no messages are published.

The logfile shows no errors:
```
[May 05 21:50:24][mqtt] MqttClient::MqttClient called
[May 05 21:50:24][mqtt] using libmosquitto 2.0.11
[...]
[May 05 21:50:24][mqtt] mqtt_client_thread created.
```

r00t-, 2024-05-08:
sadly you gave very little information,
most importantly, your vzlogger.conf and a more complete logfile.
are your meters actually working and generating readings (that could then be published)?

Coxcopi, 2024-05-09:
Hey there, thanks for your reply. The meters are generating readings regularely (about once per second), and I can see them in the logfile.
This is what the vzlogger.conf looks like:
```json
{
"retry" : 0,
"daemon": true,
"verbosity" : 15, 
"log" : "/var/log/vzlogger.log",

"local" : {
 "enabled" : false,
 "port" : 8080,
 "index" : true,
 "timeout" : 30,
 "buffer" : 600
},

"meters" : [{
 "enabled" : true,
 "protocol" : "sml",
 "device" : "/dev/ttyAMA0",
 }
],

"mqtt": {
  "enabled": true,
  "host": "192.###.###.###",
  "port": 1883,
  "id": "vzlogger",
  "cafile": "",
  "capath": "",
  "certfile": "",
  "keyfile": "",
  "keypass": "",
  "keepalive": 30,
  "topic": "vzlogger/data",
  "user": "####",
  "pass": "####",
  "retain": false,
  "rawAndAgg": false,
  "qos": 0,
  "timestamp": false
  }
}
```
I've checked the config file for syntax errors with the online config editor tool. I also tried using the mosquitto test server instead of a local mqtt server, but I wasn't able to `mosquitto_sub` any published messages either.

Here is what the full logfile looks like after starting the vzlogger service:
```
[May 09 10:00:04][main] vzlogger v0.8.3 based on heads/master-0-gd4f3b16c9f from Tue, 23 Jan 2024 21:10:05 +0100 started.
[May 09 10:00:04][mtr0] Creating new meter with protocol sml.
[May 09 10:00:04][mtr0] Meter configured, enabled.
[May 09 10:00:04]       New meter initialized (protocol=sml)
[May 09 10:00:04][mqtt] MqttClient::MqttClient called
[May 09 10:00:04][mqtt] using libmosquitto 2.0.11
[May 09 10:00:04]       Have 1 meters.
[May 09 10:00:04][main] log level is 15
[May 09 10:00:04][main] local=0
[May 09 10:00:04]       Daemonize process...
[May 09 10:00:04]       Opened logfile /var/log/vzlogger.log
[May 09 10:00:04][push] No pushDataServer defined.
[May 09 10:00:04][mqtt] mqtt_client_thread created.
[May 09 10:00:04][]     ===> Start meters
[May 09 10:00:04][mqtt] Start mqtt_client_thread
[May 09 10:00:04][mtr0] Meter connection established
[May 09 10:00:04][mtr0] Meter thread started
[May 09 10:00:04][mtr0] Meter is opened. Starting channels.
[May 09 10:00:04][]     Startup done.
[May 09 10:00:04][mtr0] Number of readers: 32
[May 09 10:00:04][mtr0] Config.local: 0
[May 09 10:00:04][mqtt] connect_callback called, res=0
[May 09 10:00:05][mtr0] Got 8 new readings from meter:
[May 09 10:00:05][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=1394002.48 ts=1715241605333
[May 09 10:00:05][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=1002.00 ts=1715241605333
[May 09 10:00:05][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=1393000.00 ts=1715241605333
[May 09 10:00:05][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=99669.37 ts=1715241605333
[May 09 10:00:05][mtr0] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=-342.91 ts=1715241605333
[May 09 10:00:05][mtr0] Reading: id=1-0:36.7.0*255/ObisIdentifier:1-0:36.7.0*255 value=-503.42 ts=1715241605333
[May 09 10:00:05][mtr0] Reading: id=1-0:56.7.0*255/ObisIdentifier:1-0:56.7.0*255 value=116.01 ts=1715241605333
[May 09 10:00:05][mtr0] Reading: id=1-0:76.7.0*255/ObisIdentifier:1-0:76.7.0*255 value=44.50 ts=1715241605333
[May 09 10:00:06][mtr0] Got 8 new readings from meter:
[May 09 10:00:06][mtr0] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=1394002.48 ts=1715241606333
[May 09 10:00:06][mtr0] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=1002.00 ts=1715241606333
[May 09 10:00:06][mtr0] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=1393000.00 ts=1715241606333
[May 09 10:00:06][mtr0] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=99669.46 ts=1715241606333
[...]
```

r00t-, 2024-05-09:
if you don't define any channels, no data will be published,
even if mqtt does not implement an "api" as it should ( https://github.com/volkszaehler/vzlogger/issues/550 ),
it only hooks into the code at a point _after_ readings have been asssociated with a channel.
see https://github.com/mbehr1/vzlogger/blob/4ddc311c19ee7558c54049b2097ef5316783864a/src/threads.cpp#L143

r00t-, 2024-05-09:
i wonder if the documentation needs to be improved.
there does not seem to be much documentation on mqtt, and no dedicated config example.
BUT the concept of channels is at the very core of vzlogger, and all over the docs and examples.
even if the current mqtt implementation bypasses some of vzlogger's logic, vzlogger is still vzlogger, and not a generic sml-to-mqtt gateway.

r00t-, 2024-05-09:
(maybe the solution to #550 is to rip the mqtt code out of vzlogger and create that generic sml-to-mqtt gateway, based in just libsml? if that's what mqtt users expect?)

closed 2024-05-09 by r00t-