volkszaehler/vzlogger #653: Integration Wasserzähler

slemke76, 2024-05-09:
Hallo,

ich habe einen M-Bus Wasserzähler (https://process.honeywell.com/us/en/products/utilities/water/digital-metering/mbus/falcon-mj-pulse-mbus) zur Verfügung. 

So wie ich das sehe, ist die "OBIS Value Group A = 8: Water" aktuell nicht implementiert? Oder irre ich?

Ich habe schon ein wenig getestet - mit den libmbus Funktionen bekomme ich den Wasserverbrauch korrekt zurück geliefert:

`root@raspberrypi:/var/log/vzlogger# mbus-serial-request-data -d -b 2400 /dev/ttyAMA3 1
init_slaves: debug: sending init frame #1
[2024-05-09 21:30:06Z] SEND (005): 10 40 FD 3D 16
[2024-05-09 21:30:06Z] RECV (001): E5
init_slaves: debug: sending init frame #2
[2024-05-09 21:30:06Z] SEND (005): 10 40 FD 3D 16
[2024-05-09 21:30:06Z] SEND (005): 10 5B 01 5C 16
[2024-05-09 21:30:07Z] RECV (087): [...]
mbus_frame_print: Dumping M-Bus frame [type 4, 87 bytes]: [...]
<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>.....</Id>
        <Manufacturer>ELR</Manufacturer>
        <Version>178</Version>
        <ProductName></ProductName>
        <Medium>Water</Medium>
        <AccessNumber>21</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (m m^3)</Unit>
        <Value>111717</Value>
        <Timestamp>2024-05-09T21:30:07Z</Timestamp>
    </DataRecord>
[...]
`

Meine Versuche in vzlogger klappen aktuell nicht, da passiert im Prinzip gar nichts. 
Ein Key für Verschlüsselung ist nicht erforderlich, ohne startet vzlogger nicht einmal.

`   "meters": [
        {
    "enabled": true,
      "allowskip": false,
      "interval": -1,
      "aggtime": -1,
      "aggfixedinterval": false,
      "protocol": "oms",
      "device": "/dev/ttyAMA3",
      "dump_file": "/var/log/vzlogger/dumpfile.log",
      "baudrate": 2400,
      "baudrate_read": 2400,
      "parity": "8n1",
      "wait_sync": "off",
      "read_timeout": 60,
      "baudrate_change_delay": 400,
      "key": "********************************",
      "mbus_debug": true,
      "use_local_time": false
   }
    ]
`

Besteht grundsätzliches Interesse, das ganze zu implementieren? Ich unterstütze gerne, habe aber akutell keinen wirklichen Ansatzpunkt.
Zugriff auf das System würde sich einrichten lassen...

Grüße
Sebasitna

r00t-, 2024-05-10:
i never used mbus so i can't comment much.
the mbus code: https://github.com/volkszaehler/vzlogger/blob/master/src/protocols/MeterOMS.cpp
has lots of debug output, so if you set loglevel to 15 and maybe also `mbus_debug` in the meter config you should get _some_ output. (you did not include a log, so i assume you did not get any in your attempts do far?)
it seems there's a conversion of mbus-specific indeitifiers to OBS,
so you probably will have to add whatever your meter is sending there:
https://github.com/volkszaehler/vzlogger/blob/master/src/protocols/MeterOMS.cpp#L349

narc-Ontakac2, 2024-05-10:
Be aware that our deb packages are unfortunately currently not build with mbus support.

slemke76, 2024-05-10:
Hello,

> so i assume you did not get any in your attempts do far?

Yes, I didn´t get anything so far.
I will take a closer look the next days - actually it is unclear to me, if the code is generally compatible with my meter. 
E.g. I do not need a key for communication.

> Be aware that our deb packages are unfortunately currently not build with mbus support.

Thanks for the hint - I compiled vzlogger on my own, "oms" is listed as supported protocol type.

Sebastian

closed 2026-05-04 by slemke76