volkszaehler/vzlogger #658: Gateway Timeout

JAK0721, 2024-05-18:
I have made a new volkszaehler/vzlogger installation on a RPI 5 (Raspberry PI OS Lite 64bit).
I followed the instruction https://wiki.volkszaehler.org/software/middleware/installation with PPM as Server (empfohlene Methode) and nginx as Proxy as well as the vzlogger installation according https://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version?s[]=vzlogger&s[]=installation

I also imported my database which was collected on a RPI 4 installation (started in 2020) and reused my vzlogger.conf.
The installation works fine (collection Power - Einspeisung und Bezug) but I always getting these anoying "Gateway Timeout" failures.
The curves on the graph are showing almost instantly, but then it takes quit long to show the total values (kWh) 

[image]

J-A-U, 2024-05-18:
> it takes quit long to show the total values (kWh).

Rebuild the aggregation table. But stop the cronjobs beforehand, it will interfere with the script running!

`php /var/www/volkszaehler.org/bin/aggregate rebuild -l day -l hour -l minute`

If this doesn't work (i might be wrong on syntax), delete and run from scratch.

`php /var/www/volkszaehler.org/bin/aggregate clear -s 1000000000 -l all`
`php /var/www/volkszaehler.org/bin/aggregate run -m full -l day -l hour -l minute`

If you are curious, you might want to check before and after: 
`http://IP-ADRESSE/middleware.php/capabilities/database.json?`

JAK0721, 2024-05-18:
Perfect. Thanks!

closed 2024-05-26 by J-A-U