volkszaehler/vzlogger #665: Test fails when builing with OMS

narc-Ontakac2, 2024-06-15:
Building the vzlogger Debian package with OMS support on Ubuntu 22.04.4 the following test failed: 
```
2024-06-15T11:54:50.8454426Z   OMS:send_frame (len=1) ok
2024-06-15T11:54:50.8454981Z ./tests/mocks/mock_MeterOMS.cpp:140: Failure
2024-06-15T11:54:50.8455601Z Expected equality of these values:
2024-06-15T11:54:50.8456146Z   rds[7].time_s()
2024-06-15T11:54:50.8456606Z     Which is: 1718452490
2024-06-15T11:54:50.8457084Z   mktime(&t)
2024-06-15T11:54:50.8457549Z     Which is: -1
2024-06-15T11:54:50.8458064Z [  FAILED  ] mock_MeterOMS.first_packets (1 ms)
2024-06-15T11:54:50.8458774Z [----------] 4 tests from mock_MeterOMS (2 ms total)
```
to be continued ...

narc-Ontakac2, 2024-06-18:
@mbehr1 Any hints on what might be happening here?

mbehr1, 2024-06-18:
hmm. weird. you closed the issue already. Is the failure understood/resolved?

narc-Ontakac2, 2024-06-18:
> Is the failure understood/resolved?

No, just clicked the wrong button.

mbehr1, 2024-06-19:
:-). I'll check this evening.

mbehr1, 2024-06-19:
hmm. I cannot reproduce on my ubuntu 22.04.4. meterOMS runs fine there. Can you point me to full logs from the failing test?

narc-Ontakac2, 2024-06-19:
Another weirdness is that the issue occurs on the Github build system ([just tried](https://github.com/volkszaehler/vzlogger/actions/runs/9585912807)) while I can not reproduce it on a local Ubuntu.

mbehr1, 2024-06-19:
or can you tell me the exact command you used to build the package?

narc-Ontakac2, 2024-06-19:
The release action just runs `debuild --no-sign`. This the calls 
```
   dh_auto_test
	cd obj-x86_64-linux-gnu && make -j4 test ARGS\+=--verbose ARGS\+=-j4
make[1]: Entering directory '/home/runner/work/vzlogger/vzlogger/obj-x86_64-linux-gnu'
Running tests...
/usr/bin/ctest --force-new-ctest-process --verbose -j4
```

I have no idea why this only happens o the build system. But I think this is something we should not just ignore.

mbehr1, 2024-06-19:
The test in GitHub actions seems to have basic problems like:
5:   s0:Reading from hardwareinterface failed with Operation not permitted. (thousands of times)

mbehr1, 2024-06-19:
Really weird. The build logs look ok. I ran valgrind on the test binary. which mainly pointed to one uninitialized var used for printf. Can be fixed with:
```
	while (expect_frame > 0) {

		mbus_frame frame = {};
``` 
in MeterOMS.cpp line 259

But I cannot see why 
a) the read value is wrong... (it's current time and not the expected time from 2015.
b) mtkime fails to convert and returns -1.

I ran a small test binary that shows no issues at all either:
```
#include <ctime>
#include <cstdio>

int main(void){
  struct tm t;
	t.tm_sec = 3;
	t.tm_min = 36;
	t.tm_hour = 23;
	t.tm_mday = 20;
	t.tm_mon = 6 - 1;
	t.tm_year = 2015 - 1900;
	t.tm_isdst = 1;
	auto r = mktime(&t);
  printf("r=%ld\n",r);
  return r >= 0 ? 0 : r;
}
```
shows no warnings, errors, no valgrind issues. 

Is there a way to download the generated test binary from the GitHub action?
in build dir: tests/mocks/mock_MeterOMS would be interesting... to debug locally with a debugger.

narc-Ontakac2, 2024-06-19:
> Is there a way to download the generated test binary from the GitHub action?

With a modified action this is doable. Not sure if this can be done without. 

A possible cause are locale or timezone settings. These may be different on the Github build system. 

But all this can be done on another day. Good night.

narc-Ontakac2, 2024-06-20:
Thinking about this broken or missing TZ data on the build system might explain it. If there is no current timezone, mktime will fail and it might be reasonable for libmbus to fall back to the current timestamp if the time from the current data can't be interpreted in a time zone context.

mbehr1, 2024-06-20:
yep! that's it. I can now repro with the small test binary from above via:
bash
export TZ=FOO
./a.out
r=-1 

:-)

So setting a TZ to a valid value should be enough?  (or in bash even `unset TZ` -> use default works)
Can you do that in the deb build?

mbehr1, 2024-06-20:
interesting that that's not mentioned in the mtkime man page.

narc-Ontakac2, 2024-06-20:
> Can you do that in the deb build?

Yes. I will also install tzdata to be sure.

narc-Ontakac2, 2024-06-20:
3 tries that did not work :-(

Any hint how I can set TZ so that it reaches the tests?

narc-Ontakac2, 2024-06-22:
@mbehr1  This is more complicated: The following should reproduce the issue, but it does not:
```
env TZ=FOO obj-x86_64-linux-gnu/tests/mocks/mock_MeterOMS 
```

mbehr1, 2024-06-22:
for me it does.

env TZ=FOO ./tests/mocks/mock_MeterOMS
```
...
Expected equality of these values:
  rds[7].time_s()
    Which is: 1719059309
  mktime(&t)
    Which is: -1
[  FAILED  ] mock_MeterOMS.first_packets (10 ms)
[----------] 4 tests from mock_MeterOMS (11 ms total)

[----------] Global test environment tear-down
[==========] 4 tests from 1 test suite ran. (11 ms total)
[  PASSED  ] 3 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] mock_MeterOMS.first_packets
```

narc-Ontakac2, 2024-06-22:
> for me it does.

Which OS +  Version?

mbehr1, 2024-06-22:
```
❯ uname -a
Linux h....stratoserver.net 4.15.0 #1 SMP Tue Jan 9 19:45:01 MSK 2024 x86_64 x86_64 x86_64 GNU/Linux
❯ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.4 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.4 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

narc-Ontakac2, 2024-06-29:
OK, I have an Ubuntu at hand now. Very weird,
` ` ` 
TZ=Etc/UCT obj-x86_64-linux-gnu/tests/mocks/mock_MeterOMS
` ` ` 
fails with a valid timezone while
` ` ` 
TZ=America/Bahia obj-x86_64-linux-gnu/tests/mocks/mock_MeterOMS
` ` ` 
succeeds.

narc-Ontakac2, 2024-06-29:
` ` ` 
TZ=Africa/Bissau obj-x86_64-linux-gnu/tests/mocks/mock_MeterOMS
` ` ` 
fails too. My current theory is that mktime does not tolerate ` tm_isdst = 1` for a time zone without dst any more.

narc-Ontakac2, 2024-06-29:
What I do not understand is why the obvious fix in ` tests/CMakeLists.txt` 
```
    ADD_CUSTOM_TARGET(test_coverage
        # Cleanup lcov
        ${LCOV_PATH} --rc lcov_branch_coverage=1 --directory . --zerocounters
        # Run tests
        COMMAND vzlogger_unit_tests
        COMMAND mock_metermap
        COMMAND mock_MeterW1therm
        COMMAND TZ="Europe/Berlin" mock_MeterOMS
        COMMAND mock_MeterS0
```
does not work.  @mbehr1 Any idea on this?

mbehr1, 2024-06-29:
> OK, I have an Ubuntu at hand now. Very weird, ` ` ` TZ=Etc/UCT obj-x86_64-linux-gnu/tests/mocks/mock_MeterOMS` ` ` fails with a valid timezone while ` ` ` TZ=America/Bahia obj-x86_64-linux-gnu/tests/mocks/mock_MeterOMS` ` ` succeeds.

there is a typo in "Etc/UCT" -> "Etc/UTC", isn't it?

narc-Ontakac2, 2024-06-29:
` ` ` 
$ timedatectl list-timezones | grep Etc/U
Etc/UCT
Etc/UTC
Etc/Universal
` ` `
It seems both are valid.

narc-Ontakac2, 2024-06-29:
Another approach to fix this would be to modify data[300] and use a different time (in winter). However I have no idea how to modify data[300].

mbehr1, 2024-06-29:
> ` ` ` TZ=Africa/Bissau obj-x86_64-linux-gnu/tests/mocks/mock_MeterOMS` ` ` fails too. My current theory is that mktime does not tolerate ` tm_isdst = 1` for a time zone without dst any more.

hmm. interesting. yes, the tm_isdst=1 does not work for those.
Well we could simply change the unit test to be
```
// ASSERT_EQ(rds[7].time_s(), mktime(&t));
ASSERT_EQ(rds[7].time_s(), 1434836163);
```
and remove all the mtkime issues (which are not part of the test anyhow)
But I just tried that and the reading itself contains a different value then either. Need to check the code.

mbehr1, 2024-06-29:
the code itself uses mtkime as well:
```
case 0x06:             // 6 byte integer (48 bit)
		if (vif == 0x6d) { // 48bit time value (CP48)
			struct tm t;
			t.tm_sec = record->data[0] & 0x3f;
			t.tm_min = record->data[1] & 0x3f;
			t.tm_hour = record->data[2] & 0x1f;
			t.tm_mday = record->data[3] & 0x1f;
			t.tm_mon = record->data[4] & 0xf;
			if (t.tm_mon > 0)
				t.tm_mon -= 1; // struct tm is 0-11 based... (months since January)
			t.tm_year =
				100 + (((record->data[3] & 0xe0) >> 5) |
					   ((record->data[4] & 0xf0) >> 1)); // tm_year is number of years since 1900.
			t.tm_isdst = ((record->data[0] & 0x40) == 0x40) ? 1 : 0;
			// check for time invalid at bit 16 (1-based)
			if ((record->data[1] & 0x80) == 0x80) {
				// time invalid!
			} else {
				print(log_finest, "time=%.2d-%.2d-%.2d %.2d:%.2d:%.2d", name().c_str(),
					  1900 + t.tm_year, t.tm_mon + 1, t.tm_mday, t.tm_hour, t.tm_min, t.tm_sec);
				// convert to double (as seconds since 1970-01-01
				toRet = mktime(&t);
			}
		} 
```
and here the is_dst is coming from the received data.
(so mtkime should not fail...)
We could add a workaround that if mktime parsing fails, try without.
On the other side the meter might not return that value in a timezone where isdst is not valid...
But I'd still prefer a better mechanism.

mbehr1, 2024-06-29:
Hmm. What seems to work is:
```
case 0x06:             // 6 byte integer (48 bit)
		if (vif == 0x6d) { // 48bit time value (CP48)
			struct tm t;
			t.tm_sec = record->data[0] & 0x3f;
			t.tm_min = record->data[1] & 0x3f;
			t.tm_hour = record->data[2] & 0x1f;
			t.tm_mday = record->data[3] & 0x1f;
			t.tm_mon = record->data[4] & 0xf;
			if (t.tm_mon > 0)
				t.tm_mon -= 1; // struct tm is 0-11 based... (months since January)
			t.tm_year =
				100 + (((record->data[3] & 0xe0) >> 5) |
					   ((record->data[4] & 0xf0) >> 1)); // tm_year is number of years since 1900.
			t.tm_isdst = -1; //  ((record->data[0] & 0x40) == 0x40) ? 1 : 0;
```
so setting there (and in the test as well). the tm_isdst to -1 (auto detect at mktime).
I tried here quickly with e.g. setting it to -1 in MeterOms and keeping it at 1 in the test and that seems to work (so imho not changing the MeterOMS behavior at least for our timezone.

narc-Ontakac2, 2024-06-29:
> (so mtkime should not fail...)

My theory is that mktime fails if tm_isdst is set while a time zone without DST is used. So this will should too.

narc-Ontakac2, 2024-06-29:
But seeing the above I should be able to move the time to winter.

narc-Ontakac2, 2024-06-29:
> keeping it at 1 in the test and that seems to work

That is surprising, I would expect this to fail, too.

narc-Ontakac2, 2024-06-29:
Thinking about this ignoring an isdst flag in the data is a questionable move. There might be old meters that do not have DST. It might however be the only way if mktime does not tolerate a deviating isdst flag.

mbehr1, 2024-06-29:
we could as well a logic like:
- determine at startup whether a is_dst of 1 is supported.
- if yes -> use it
- if no -> fallback to -1

But I wonder whether the (use -1 always) makes even more sense.

narc-Ontakac2, 2024-06-29:
> But I wonder whether the (use -1 always) makes even more sense.

It is simpler, I would use that. If anybody reports a bug we can still go for a more complex solution.

mbehr1, 2024-06-29:
do you change it or shall I add create a PR?

narc-Ontakac2, 2024-06-29:
I will, I'll also undo my latest try.

narc-Ontakac2, 2024-06-29:
Build succeeded.

closed 2024-06-30 by narc-Ontakac2