volkszaehler/vzlogger #677: One meter for aggregated influx logging AND non aggregated local for EVCC

grasshide, 2024-08-31:
My setup consists of one Smartmeter with a SML interface. My goal is to have the 1.8.0 counter send every five minutes to influx. And additionally I want the 16.7.0 consumption to be available on the local (http) interface as non aggregated/live for EVCC.
The influx part works as expected with my config. Only the 1.8.0 value is logged.

But on the http interface all values are aggregated:
`{ "version": "0.8.7", "generator": "vzlogger", "data": [ { "uuid": "1", "last": 1725095946119, "interval": -1, "protocol": "sml", "tuples": [ [ 1725095908120, 3050.5891000000001 ] ] }, { "uuid": "2", "last": 1725095946119, "interval": -1, "protocol": "sml", "tuples": [ [ 1725095908120, 3.9500000000000002 ] ] } ] }`

This is my config:
`{
"retry" : 0,
"daemon": true,
"verbosity" : 1,
"log" : "/var/log/vzlogger.log",

"local" : {
 "enabled" : false,
 "port" : 8088,
 "index" : true,
 "timeout" : 30,
 "buffer" : -1

},

"local": {
    "enabled": true,
    "port": 8087,
    "index": true,  
    "timeout": 30,
    "buffer": -1
},

"meters" : [
  {
  "enabled" : true,
  "protocol" : "sml",
  "device" : "/dev/ttyAMA0",
  "aggtime": 300,
  "channels": [
    {
      "api": "influxdb",
      "uuid": "1",
      "identifier" : "1-0:1.8.0",
      "host": "192.168.1.2:8087",
      "token": "my_token",
      "organization": "my_org",
      "aggmode": "max"
    },
    {
      "api": "null",
      "uuid": "2",
      "identifier": "1-0:16.7.0",
      "aggmode": "none"
    }
    ],
 }
]}`

I tryied to configure two meters, but that doesn't work since I can't read the tty buffer twice.
What do I have to change to get the "raw" value on the local interface?

r00t-, 2024-09-02:
https://github.com/volkszaehler/vzlogger/issues/609#issuecomment-1793538358

r00t-, 2024-09-07:
duplicate of #609

closed 2024-09-07 by r00t-