volkszaehler/vzlogger #678: adduser problem - Package installation via cloudsmith

kmecke, 2024-10-02:
After the package installation via cloudsmith as described in the Readme in section "Debian and Raspberry Pi OS Packages" everything works out nice, until the configuration of the package. I think there is just the option missing so that the user _vzlogger user will be created correctly. 

```
vz2@rpi:~ $ sudo apt install vzlogger
[...]
Get:4 https://dl.cloudsmith.io/public/volkszaehler/volkszaehler-org-project/deb/raspbian bullseye/main armhf vzlogger armhf 0.8.7 [251 kB]
Fetched 476 kB in 1s (325 kB/s)
Selecting previously unselected package libmbus0.
(Reading database ... 53780 files and directories currently installed.)
Preparing to unpack .../libmbus0_0.9.0+git20240530-1_armhf.deb ...
Unpacking libmbus0 (0.9.0+git20240530-1) ...
Selecting previously unselected package libmicrohttpd12:armhf.
Preparing to unpack .../libmicrohttpd12_0.9.72-2+deb11u1_armhf.deb ...
Unpacking libmicrohttpd12:armhf (0.9.72-2+deb11u1) ...
Selecting previously unselected package libsml1:armhf.
Preparing to unpack .../libsml1_1.1.3_armhf.deb ...
Unpacking libsml1:armhf (1.1.3) ...
Selecting previously unselected package vzlogger.
Preparing to unpack .../vzlogger_0.8.7_armhf.deb ...
Unpacking vzlogger (0.8.7) ...
Setting up libmbus0 (0.9.0+git20240530-1) ...
Setting up libsml1:armhf (1.1.3) ...
Setting up libmicrohttpd12:armhf (0.9.72-2+deb11u1) ...
Setting up vzlogger (0.8.7) ...
adduser: Please enter a username matching the regular expression configured
via the NAME_REGEX configuration variable.  Use the `--force-badname'
option to relax this check or reconfigure NAME_REGEX.
dpkg: error processing package vzlogger (--configure):
 installed vzlogger package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u11) ...
Errors were encountered while processing:
 vzlogger
E: Sub-process /usr/bin/dpkg returned an error code (1)

vz2@rpi:/etc $ cat os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
```

My workaround, which does not work (but why?):
```
sudo nano /etc/adduser.conf
NAME_REGEX="^[a-z_][-a-z0-9_]*\$"
```

Maybe you want to change the username OR add the --force-badname

... and thank you for your great work at this project!

J-A-U, 2024-10-02:
Can you generate the needed user on the command line?
`sudo useradd -M vzlogger`

kmecke, 2024-10-02:
Hi -  yes that works fine.... 
Having said that... is the underscore the issue? I understood that the service user is _vzlogger? As mentioned [here](https://github.com/volkszaehler/vzlogger/blob/507f55db41a2415b5f45ba994af79980524cb9b0/debian/changelog#L23).

I just created the _vzlogger user as well, without issues. So I guess the modified regex pattern in adduser.conf has some effect.

Surprise.. here wo go:
```
vz2@rpi:/usr/local/bin $ sudo dpkg --configure -a
Setting up vzlogger (0.8.7) ...
Created symlink /etc/systemd/system/multi-user.target.wants/vzlogger.service → /lib/systemd/system/vzlogger.service.
vz2@rpi:/usr/local/bin $
```

So I guess the workaround works now.

J-A-U, 2024-10-02:
> As mentioned [here](https://github.com/volkszaehler/vzlogger/blob/507f55db41a2415b5f45ba994af79980524cb9b0/debian/changelog#L23).

Sorry. Missed that one.

> So I guess the workaround works now.

Did you delete the created users before you retried installing the package?

If not, my bet is the needed user already existing "solved" the problem.

kmecke, 2024-10-03:
Hi - no, I did not delete the user. I think you are right. I feel like I am pretty close to get my volkszähler up an running... this weekend.

Shall I change the pattern back to the original expression too reproduce the issue in the installation process, or do you have enough information? I think its a pretty small thing that breaks the package installation. Let me know if I can assist you...

In the meantime, I am a happy customer by adding the user manually ; ) 
I leave the ticket open and up to you how to proceed. Thanks for your support and your great work!

Zugschlus, 2024-10-04:
adduser maintainer from Debian here. Yes, to create a user with an underscore you need --force-badname, as mentioned in the error message.

J-A-U, 2024-10-04:
@Zugschlus is there a best practice how to name "system"-users (without home directory) for Debian?

r00t-, 2024-10-05:
this was introduced by https://github.com/volkszaehler/vzlogger/commit/8e28a5c61f27bbbc33f24105e197b4a9c406b136
it looke like the underscore username was recommended to @narc-Ontakac2 as a best practice...
@narc-Ontakac2: can you comment?

r00t-, 2024-10-05:
as quoted in #628 :
https://www.debian.org/doc/debian-policy/ch-opersys.html#users-and-groups
> When maintainers choose a new hardcoded or dynamically generated username for packages to use, they should start this username with an underscore. This minimizes collisions with locally created user accounts.

this appears to entail that scripts generating such an account have to use `--force-badname`, even if that seems strange.

narc-Ontakac2, 2024-10-05:
I have asked about [the issue on the Debian developer list](https://lists.debian.org/debian-devel/2024/ 10/msg00035.html). I'll fix this according to the outcome.

narc-Ontakac2, 2024-10-05:
The script is creating a system user. Therefore 
```
# System user- and groupnames are checked against this regular
# expression.
# Default: SYS_NAME_REGEX="^[a-z_][-a-z0-9_]*\$?$"
```
should be applied. This however _allows_ a leading underscore.  The above is however from Debian trixie.
@kmecke What is your /etc/adduser.conf?

r00t-, 2024-10-06:
https://metadata.ftp-master.debian.org/changelogs//main/a/adduser/adduser_3.134_changelog
> adduser (3.131) unstable; urgency=medium
>  * Fix a bug: system user names are now allowed to begin with an
>    underscore. Thanks to Guillem Jover for spotting this.
>  -- Marc Haber <mh+debian-packages@zugschlus.de>  Tue, 07 Feb 2023 12:08:36 +0100

https://packages.debian.org/search?keywords=adduser&searchon=names&suite=all&section=all
> buster (oldoldstable)   3.118: all
> bullseye (oldstable)    3.118+deb11u1: all
> bookworm (stable)      3.134: all
> trixie (testing)              3.137: all
> sid (unstable)              3.137: all

so this should work on stable and beyond, and only break on oldstable.
but @kmecke is installing on bullseye/oldstable. as of his log...
if we want to support that, we need to add the force flag.
i guess just adding it make more sense than special-casing on the distribution or adduser-version.

kmecke, 2024-10-06:
Even though you, @r00t- have it all together, I still want to answer your @narc-Ontakac2 question, maybe 8 hrs late... 

> @kmecke What is your unmodified /etc/adduser.conf?

The regex line was a comment and the regex was missing the underscore. Last two lines:  

```
# check user and group names also against this regular expression.
# NAME_REGEX="^[a-z][-a-z0-9_]*\$"
```
I felt reluctant to upgrade to stable, since lots of stuff on that Raspi 3B (PiHole, z-wave controller... soon volkszähler) seem to be tested toward bullseye. But you are right - its about time.  Stable should stable enough for an dist-upgrade

narc-Ontakac2, 2024-10-06:
Unfortunately the trixie man page for adduser states 
```
       --force-badname
       --allow-badname
              These are the deprecated forms of --allow-bad-names.  It will  be
              removed  during  the  release  cycle  of the Debian release after
              bookworm.
```
So special-casing is needed unless we bet on the adduser maintainers being sloppy regarding their announcement.

r00t-, 2024-10-06:
@Zugschlus:
as you are here anyway, do you have an idea of how this should be handled for packages that want to support old and new releases?

narc-Ontakac2, 2024-10-07:
There is [PR #679](https://github.com/volkszaehler/vzlogger/pull/679) by now, I do however still need to test it.

narc-Ontakac2, 2024-10-10:
@kmecke In case you haven't upgraded yet - could you remove vzlogger, do a `deluser _vzlogger` and install it again. The issue should be fixed in 0.8.8. However I haven't been able to test it.

kmecke, 2024-10-10:
Hi @narc-Ontakac2 - sorry, I did the upgrade (vzloggger still not running, but the user is not the issue I guess). I have a backup of the old system, but I will not find the time to restore it and test it soon (maybe in two weeks)...

closed 2024-10-11 by narc-Ontakac2