volkszaehler/vzlogger #680: Dockerfile is not functional anymore

ako-3004, 2024-11-24:
Docker build quits with error "The source directory "/vzlogger" does not appear to contain CMakeLists.txt", but file is there.

` => ERROR [builder 7/7] RUN cmake -DBUILD_TEST=off     && make -j $(nproc --all || echo 1)     && make install     && if [ "off" != "off" ]; then make test; fi     0.3s
------
 > [builder 7/7] RUN cmake -DBUILD_TEST=off     && make -j $(nproc --all || echo 1)     && make install     && if [ "off" != "off" ]; then make test; fi:
0.263 CMake Warning:
0.263   No source or binary directory provided.  Both will be assumed to be the
0.263   same as the current working directory, but note that this warning will
0.263   become a fatal error in future CMake releases.
0.263
0.263
0.263 CMake Error: The source directory "/vzlogger" does not appear to contain CMakeLists.txt.
0.263 Specify --help for usage, or press the help button on the CMake GUI.
`

r00t-, 2024-11-28:
please include the exact invocation you are using to build.
most likely you are using the dockerfile incorrectly.
it expects to be run with an existing checkout as the build context,
it copies the source into the container using
`COPY . /vzlogger`.
if you run a copy of the dockerfile without the build context, it will fail. as you describe.

a working invocation is:
```
$ cd /tmp/ && git clone git@github.com:volkszaehler/vzlogger.git && cd vzlogger && docker build .
```

r00t-, 2024-11-28:
duplicate of https://github.com/volkszaehler/vzlogger/issues/619 too.
will consider adding some code to the dockerfile to catch this (ab)use and generate an explicit error message.

closed 2024-11-28 by r00t-

r00t-, 2024-11-28:
```
[/vzlogger]$ docker build - <Dockerfile
Sending build context to Docker daemon  3.584kB
Step 1/18 : FROM alpine:latest as builder
 ---> 28f6e2705743
Step 2/18 : COPY . /vzlogger
 ---> 49fdd3c1cd71
Step 3/18 : COPY CMakeLists.txt /vzlogger/
COPY failed: file not found in build context or excluded by .dockerignore: stat CMakeLists.txt: file does not exist
```
```
[vzlogger]$ docker build .
(succeeds)
```