volkszaehler/vzlogger #683: Updated to current master and install script is not running any longer

dm82m, 2024-12-05:
1)
```
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 3 (delta 0), reused 0 (delta 0), pack-reused 0 (from 0)
Unpacking objects: 100% (3/3), 4.06 KiB | 159.00 KiB/s, done.
From https://github.com/volkszaehler/vzlogger
   b936b96..6adcad8  master     -> origin/master
Updating b936b96..6adcad8
Fast-forward
 install.sh | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
```

2)
```
pi@raspberrypi:~/vzlogger $ ./install.sh vzlogger
vzlogger installation script

checking prerequisites: grep
 extrace: not found. Please install to use this script (e.g. sudo apt-get install extrace).
 For complete list of build requirements please check http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version
```

3)
```
pi@raspberrypi:~/vzlogger $ sudo apt-get install extrace
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  extrace
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 12.8 kB of archives.
After this operation, 42.0 kB of additional disk space will be used.
Get:1 http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian bookworm/main armhf extrace armhf 0.7-2 [12.8 kB]
Fetched 12.8 kB in 0s (34.7 kB/s)  
Selecting previously unselected package extrace.
(Reading database ... 108312 files and directories currently installed.)
Preparing to unpack .../extrace_0.7-2_armhf.deb ...
Unpacking extrace (0.7-2) ...
Setting up extrace (0.7-2) ...
Processing triggers for man-db (2.11.2-2) ...
```

4)
```
pi@raspberrypi:~/vzlogger $ ./install.sh vzlogger
vzlogger installation script

checking prerequisites: grep extrace git cmake pkg-config
 dh-autoreconf: not found. Please install to use this script (e.g. sudo apt-get install dh-autoreconf).
 For complete list of build requirements please check http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version
```

5)
```
pi@raspberrypi:~/vzlogger $ sudo apt-get install dh-autoreconf
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dh-autoreconf is already the newest version (20).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

happy for any advice here!

justinotherguy, 2024-12-05:
Hi @dm82m ,
thanks for your report!
What os and what platform are you running? (lsb_release -a)

dm82m, 2024-12-05:
```
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

justinotherguy, 2024-12-05:
thx!
Can you please try the install.sh from https://github.com/volkszaehler/vzlogger/tree/justin-update-readme-bookworm ?

Before starting the script, please use the updated list of required packages:
```
sudo apt-get install extrace git-core cmake libtool dh-autoreconf uuid-dev libmicrohttpd-dev \
    build-essential pkg-config libgnutls28-dev libssl-dev libmosquitto-dev libgcrypt20-dev libcurl3-dev \
    libunistring-dev libssl-dev libsasl2-dev libcurl4-openssl-dev uuid-dev libtool libltdl-dev
```

dm82m, 2024-12-05:
alle packages were already installed and up2date. with the install.sh of your link it is back working and building ...