volkszaehler/vzlogger #686: aggregated and raw values in parallel

indus, 2024-12-19:
would like to write aggregated values (```“aggtime”: 60```) to an Influx database and at the same time retrieve live data (```“aggtime”: -1```) via the local HTTP service (httpd).

I thought I could simply specify the same device again in the configuration with a different ```aggtime```, but unfortunately this is not possible. Maybe I'm doing something wrong in the config? Or is there another possibility?

My config looks like this:

``` JSONC
{
  "retry": 0,
  "daemon": true,
  "verbosity": 3,
  "local": {
        "enabled": true,
        "port": 8081,    
        "index": true,   
        "timeout": 30,   
        "buffer": -1                      
  },
  "meters": [
    {
      "enabled": true,
      "skip": true,
      "aggtime": 30,
      "protocol": "sml",
      "device": "/dev/eHZ0",
      "channels": [
        {
          "api": "influxdb",
          "uuid": "a5d64b0c",
          "identifier": "1-0:16.7.0",
          "measurement_name": "eHZ0_167",
          "aggmode": "avg",
          "host": "http://172.17.0.1:8086",
          "database": "home",
          "organization": "my_org",
          "token": "my_token",
        },
        {
          "api": "influxdb",
          "uuid": "a5d64b00",
          "identifier": "1-0:1.8.0",
          "measurement_name": "eHZ0_180",
          "aggmode": "max",
          "host": "http://172.17.0.1:8086",
          "database": "home",
          "organization": "my_org",
          "token": "my_token",
        }
      ]
    }
/* *** this doesn't work ***
    ,{
      "enabled": true,
      "skip": true,
      "aggtime": -1,
      "protocol": "sml",
      "device": "/dev/eHZ0",
      "channels": [
        {
          "api": "null",
          "uuid": "8c10862c",
          "identifier": "1-0:16.7.0",
          "measurement_name": "eHZ0_167_live",
          "aggmode": "none"
        }
      ]
    }
*/
  ]
}
```

r00t-, 2024-12-19:
this is a duplicate of an issue that is on the first page of the tracker....
https://github.com/volkszaehler/vzlogger/issues/609#issuecomment-1793538358
there is a suggested workaround there, why don't you try it and report results?

indus, 2024-12-19:
sorry - must have missed it. Testing this properly may not be feasible due to my lack of cpp knowledge. But thanks for your fast response.

r00t-, 2024-12-20:
closing as duplicate of #609

closed 2024-12-20 by r00t-