volkszaehler/vzlogger #690: Fails to build with CMake 4

narc-Ontakac2, 2025-09-01:
See [FTBFS with CMake 4](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1113626).

We'll have to take care of this when CMake for reaches unstable.

r00t-, 2025-09-04:
so the error is this:
https://debusine.debian.net/debian/developers/artifact/XXXX
```
CMake Error at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
```

so it seems we must first bump the declared compatibility to 4.0 and THEN check if anything _actually_ breaks.

narc-Ontakac2, 2025-09-06:
Thanks for the research. If it is trivial I can simply build it locally for now. There is another one with gcc15:
```
#error C++ versions less than C++17 are not supported.
```
I think both can be handled in one ticket, so I changed the title.

r00t-, 2025-09-06:
i dislike both of those because they force us to declare higher version requirements without actually making any code changes, using new features, or using features that are being removed..
it's the opensource version of planned obsolescence....
i'd prefer to put those declarations behind conditional that only enable them when actually using the newer version that demands them, but it's probably more work than it's worth.

narc-Ontakac2, 2025-09-07:
I can put them in a Debian specific patch.

narc-Ontakac2, 2025-09-09:
There is a Debian tool named pbuilder that does a build in a VM with a selected distro. I can do that next week. This week I can't.

The build works on trixie after removing 
```
 project(vzlogger)
-cmake_minimum_required(VERSION 2.6.2 FATAL_ERROR)
 if(POLICY CMP0011)
```
and
```
     set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Winit-self -Wmissing-include-dirs -Wno-pragmas -Wredundant-decls")
-    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-unused-parameter -std=c++14")
+    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-unused-parameter")
     set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fpermissive")
```
I'll test that on unstable next week.

narc-Ontakac2, 2025-10-08:
Debian is not happy about this, too: https://lists.debian.org/debian-devel/2025/10/msg00099.html