volkszaehler/vzlogger #694: vzlogger is eating memory when a channel is not reachable

chehrlic, 2025-09-21:
For testing purposes I added a channel (api: volkszaehler) but forgot to remove it afterwards. Everything worked fine until I saw that vzlogger needed more and more ram. I eventually restarted vzlogger every month or so. During restart (systemctl restart) the shutdown did not work well and I had to kill the old instance. 
I finally found some time to investigate it further and there I found my old channel entries. I removed them and now vzlogger is running fine without taking more and more memory over time.
Looks like the requests which could not be delivered are queued up somewhere without an upper limit.

claashk, 2026-01-11:
> It's been a very long time since I've worked on the API, but if I recall correctly the InfluxDB implementation should buffer values(?)

The InfluxDB API does not delete Readings it could not submit from the channel buffer, that is correct. However, during the next aggregation (e.g. using a max operation), these values will be lost. The Volkszähler API copies from the channel buffer into a transfer buffer and then clears the channel buffer. The transfer buffer is only cleared after successful transfer. This is the more stable solution, in my opinion and thus I implemented this logic in terms of a separate class ([TransferBuffer](https://github.com/claashk/vzlogger/blob/unified_api_buffering/include/TransferBuffer.hpp)), which is then used by both APIs.

While implementing this, I realized that all available APIs share a lot of code. It might be worth considering, whether code duplication could be further reduced, e.g. by creating a common HTTP-API base class, from which all current APIs could derive.

And as an additional comment regarding memory footprint (as original topic of this question):  I think it should be possible to reduce the size of a Reading, since a tuple of time and value would suffice almost everywhere (128 bit). The delete / undelete functionality is only required in very few places and could thus be separated from the Reading struct (e.g. in terms of a std:vector<bool>).

claashk, 2026-04-25:
Yes, I have been using this without problems since 4 months now. The code is here:
https://github.com/claashk/vzlogger/tree/unified_api_buffering

claashk, 2026-05-02:
With the previously mentioned defaults for max buffer size you should have a buffer of 4500 * 100 = 450000 messages. If you log one message every 30 seconds, this amounts to roughly 156 days of buffer capacity.